Klipper3d/klipper #2753: LPC17xx Questions - I2C pins and disable LEDs

dushyantahuja, 2020-04-20:
Not an issue as such - just a few questions - I have a Cohesion3D ReMix board - a Smoothieboard derivative (attached pinout), and am looking to use the full features of the board:

1. The LPC board has two I2C ports, and as far as I could see in the C code, only the first one is defined (pins 0.0 and 0.1). Is it possible to define the second one as well - (0.27 and 0.28)? 
2. The sixth stepper on the board uses Pins 1.19, 1.20 and 1.21 - these pins are also connected to the LEDs on the board. Smoothieware allows you to use these pins by setting leds_disable to true in the config (http://smoothieware.org/lpc1769-pin-usage) - Is something similar possible in Klipper?

Thanks for the great work
[image]
.

KevinOConnor, 2020-04-21:
> The LPC board has two I2C ports, and as far as I could see in the C code, only the first one is defined (pins 0.0 and 0.1). Is it possible to define the second one as well - (0.27 and 0.28)?

Yes, it's possible.  It requires some C micro-controller code changes, but they are not difficult.  I can provide some sample code if you have the hardware ready to test it.

> The sixth stepper on the board uses Pins 1.19, 1.20 and 1.21 - these pins are also connected to the LEDs on the board. Smoothieware allows you to use these pins by setting leds_disable to true in the config (http://smoothieware.org/lpc1769-pin-usage) - Is something similar possible in Klipper?

If you want to use those pins as a stepper, then you can use them - no special logic needed.  (If using the config/generic-smoothieboard.cfg as an example, just be sure to not define those pins in the `[static_digital_output leds]` section.)

If you want to use them as LEDs that are controllable at run-time, then you can define `[output_pin myled]` config sections and use the SET_PIN g-code command to control them - see config/example-extras.cfg for the details.

You can't define the pins as LEDs and steppers at the same time.

-Kevin

dushyantahuja, 2020-04-21:
Thanks Kevin.

I have the board ready and can test the second I2C port whenever you are
ready.

Will try to run the the stepper - my main concern is that these pins also
control the onboard LEDs by default. They switch on when the board goes on,
even though I don't have any LEDs section in my config.

dushyantahuja, 2020-04-21:
Tried to define extruder1 to use the pins for the sixth stepper mentioned above - but the extruder didn't work. Please see attached klippy.log

[extruder1]
max_extrude_only_distance: 100.0
step_pin: P1.20
dir_pin: P1.19
enable_pin: !P1.21
step_distance: 0.000895522
filament_diameter: 1.750
shared_heater: extruder
nozzle_diameter: 0.40
pressure_advance: 0.01 #0.0012
pressure_advance_smooth_time: 0.040
pressure_advance_lookahead_time: 0.010

[klippy.log](https://github.com/KevinOConnor/klipper/files/4511736/klippy.log)

As mentioned, my guess is that since these pins are switching on the on-board LEDs - they're unable to address the steppers.

KevinOConnor, 2020-04-21:
> I have the board ready and can test the second I2C port whenever you are ready.

I added a work-lpc176x-20200421 test branch: `cd ~/klipper ; git fetch ; git checkout origin/work-lpc176x-20200421 ; sudo service klipper stop ; make ; make flash ; sudo service klipper start`

> As mentioned, my guess is that since these pins are switching on the on-board LEDs - they're unable to address the steppers.

If the board has some kind of external mux that selectively routes those pins, then you'll need to identify the mux and how it is controlled.  The info is typically in the board schematics, but may also be found in the smoothiware code.

-Kevin

dushyantahuja, 2020-04-21:
Tried to connect an ssd1306 LCD using the pins and received an i2c timeout error.

[display]
lcd_type:ssd1306
i2c_bus: i2c0
display_group:_default_16x4

Not sure what I'm doing wrong. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4512224/klippy.log)

KevinOConnor, 2020-04-21:
FYI, I found a typo in the test code.  I recommend you retry: `cd ~/klipper ; git fetch ; git checkout origin/work-lpc176x-20200421 ; sudo service klipper stop ; make ; make flash ; sudo service klipper start`

-Kevin

dushyantahuja, 2020-04-21:
Works perfectly now - thanks a lot.

On the stepper pins - was looking at the Smoothieware code, and it seems that setting leds_disable to true only switches off the leds in the software - there is no hardware mux:

https://github.com/Smoothieware/Smoothieware/blob/0faa088fe1a2207f6c0b99ec7abccfbd1162f730/src/main.cpp#L231

if(kernel->is_using_leds()) {
        // set some leds to indicate status... led0 init done, led1 mainloop running, led2 idle loop running, led3 sdcard ok
        leds[0]= 1; // indicate we are done with init
        leds[3]= sdok?1:0; // 4th led indicates sdcard is available (TODO maye should indicate config was found)
    }

I'm not sure where in the microcontroller code, these LEDs are switched on.

KevinOConnor, 2020-04-21:
Okay, I pushed the i2c changes to the master branch (commit 8ac81fed).

As for the LEDs, if you can obtain the board schematic, that would probably help.  Out of curiosity, does the board actually have these LEDs (that is, do they flash when trying to move the stepper)?

-Kevin

dushyantahuja, 2020-04-22:
Thanks Kevin - please find attached the eagle schematic for the board. The board has LEDs connected to these pins, and they switch on when trying to move the stepper, but don't flash as they should - I'm guessing that there is some other code that is switching these on as well.  
[C3D ReMix Design Files.zip](https://github.com/KevinOConnor/klipper/files/4513839/C3D.ReMix.Design.Files.zip)

KevinOConnor, 2020-04-24:
FYI, I don't have an eagle file viewer handy.  I don't know of any reason why Klipper would not be able to control the stepper motor - even if they are also connected to leds.

-Kevin

dushyantahuja, 2020-04-24:
Hmm - I think these pins are also being controlled directly in the microcontroller code - as they switch on when the board switches on, and that might be conflicting with the stepper code. I'm not sure where I would find the code that switches on the LEDs.

KevinOConnor, 2020-04-24:
Klipper doesn't set any pins unless told to (via make menuconfig or via the printer.cfg file).  It's possible the board bootloader is setting the pins, but I doubt that would be a problem even if it did.

-Kevin

dushyantahuja, 2020-05-08:
I'm going to close this issue - I'm guessing I managed to burn out the pins, or that they are just not usable. 

Setting 
```[output_pin DIR6]
pin:P1.19
``
And using `SET_PIN PIN=DIR6 VALUE=1` they are not lighting up. I'm sure I had seen them light up previously. I'll probably try to install Smoothieware some day to test. Till then, I can't even test them anymore.

closed 2020-05-08 by dushyantahuja