Klipper3d/klipper #2754: [feature] No gcode to ask for homed axis ?

proto3, 2020-04-20:
Maybe I'm wrong, but I see no clean way to ask Klipper if axes are homed or not. The only one I know is issuing a G1 and see if we get an error like `!! Must home axis first: 0.000 0.000 2.000 [0.000]`

A solution could be making this appear in GET_POSITION response. In #2749, I propose to replace position by "Unhomed" in kinematic field :
`GET_POSITION`
`// mcu: stepper_x:0 stepper_y:0 stepper_z:0`
`// stepper: stepper_x:0.000000 stepper_y:0.000000 stepper_z:0.000000`
**`// kinematic: X:Unhomed Y:Unhomed Z:Unhomed`**
`// toolhead: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000`
`// gcode: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000`
`// gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000`
`// gcode homing: X:0.000000 Y:0.000000 Z:0.000000`
`ok`

Otherwise, adding another extended command like "GET_HOMING_STATE" would do the trick.

mental405, 2020-04-20:
You can get the homed axes via the printer object in a gcode macro

 `{% if 'xy' in printer.toolhead.homed_axes %}`

for instance.
or 

`[gcode_macro GET_HOMING_STATE]
gcode:
   {% if 'x' in printer.toolhead.homed_axes %}
      { printer.gcode.action_respond_info('X: Homed') }
   {% else %}
      { printer.gcode.action_respond_info('X: Not Homed') }
   {% endif %}`

I use this regularly in my homing macros to determine if I need to do hops or to prevent doing extra homing moves that don't need doing at that time.

ld3300, 2020-04-21:
Well, I can't test it right now because my printer is busy, but I put something together that may work.  It is just a few changes starting about line 634 in gcode.py.  You are welcome to give it a try and see if it is close or crashes and burns.  You could either download the [repo](git@github.com:ld3300/klipper.git), or just do a diff with master and copy the few changes (should be simple, plus you may see a mistake. If all checks out I'll test it on my end too later then submit it.

KevinOConnor, 2020-05-11:
I'm going to close this as it looks like the original question was answered (one can create their own g-code command to obtain the status).

-Kevin

closed 2020-05-11 by KevinOConnor