Klipper3d/klipper #2755: TMC doesn't move the motors

Mckevinho, 2020-04-20:
Hi,

A week ago i started with using klipper. After spending a lot of time to install and setup klipper with an old MKS gen L (8bit) board with DRV8825 drivers for testing.
I got klipper ready to use
:D
But i have a new problem.
After it works with the MKS board, i changed the board into the SKR1.4 LPC1768 with TMC2130 SPI drivers.
I changed the config file but now the stepper motors do nothing anymore :(
Before, the skr was correctly working with marlin so the wiring is correctly.

Everytime when i gave the home command, he gave me the homing duration fail error and i must restart the klipper firmware on the raspberry pi and also push the restart button on the skr board 

i've uploaded the klippy.log below

hopely someone knows the solution for this issue.
Kevin

[klippy.log](https://github.com/KevinOConnor/klipper/files/4505970/klippy.log)

KevinOConnor, 2020-04-21:
You need to define the correct spi pins for the tmc2130.  I updated the config/generic-bigtreetech-skr-v1.4.cfg with that info (commit 0b05a383).  Grab the latest code (see the FAQ for upgrade instructions) and use the comments in that example file to define the tmc2130 drivers.

-Kevin

Mckevinho, 2020-04-24:
Hi,

Thx Kevin, now the stepper motors work fine :D

KevinOConnor, 2020-05-02:
Great, lets close this then.

-Kevin

closed 2020-05-02 by KevinOConnor