Klipper3d/klipper #2756: Endstop phase calibration: "Stats not available..."

Hairyloon, 2020-04-20:
I discovered the idea of endstop phase calibration through the novel process of looking in the manual: https://github.com/KevinOConnor/klipper/blob/master/docs/Endstop_Phase.md

Sounds clever and a very good idea, so I've tried following the instructions, but it just gives me: 
Send: ENDSTOP_PHASE_CALIBRATE STEPPER=stepper_z
Recv: !! Stats not available for stepper stepper_z
It may be that I have done something wrong, or not taken enough data points, or it may be that I have the wrong sort of motors. I have no way of knowing which it is, or if it is something different.

Hairyloon, 2020-04-25:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4533484/klippy.log)

KevinOConnor, 2020-05-02:
The endstop phase support requires using trinamic drivers, or it has to be manually configured as described in the documentation.

-Kevin

Hairyloon, 2020-05-02:
To say "This parameter must be provided." hardly describes how to do it.
A brief word about WTF is a "Trinamic driver" wouldn't go amiss either.

closed 2020-05-11 by KevinOConnor

Hairyloon, 2020-05-11:
Should I take that to mean that Klipper is intended as a "Not for newbies" project?
It's your baby so that's entirely up to you, but IMO it is too good a concept to keep to an exclusive clique.