Klipper3d/klipper #2757: BED_MESH_CALIBRATE starts a little closer to bed than G28 home

marcosscriven, 2020-04-21:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4511222/klippy.zip)

I'm using `BED_MESH_CALIBRATE` but found after the process the nozzle was just ~0.2mm higher after homing than while calibrating.

I did the following to diagnose:

1. Home all axes - moves to 0,0,0 according to end stops.
2. Issue `BED_MESH_CALIBRATE` - now above first screw
3. Issue `ABORT`
4. Observe height of nozzle above bed
5. Home just Z axis - this result in immediately hitting end stop, pulling back, then slowly moving down again
6. Observe that nozzle is slightly closer than beginning of process
7. Issue `BED_MESH_CALIBRATE` again
8. Watch as nozzle moves towards bed and is slightly nearer again.

My theory is that homing by GCODE means it approaches a second time more slowly, and therefore gives the end time slightly more time to respond.

So I guess the question is, how do I get `BED_MESH_CALIBRATE` to home with the same profile as GCODE?

KevinOConnor, 2020-04-24:
If you run G28, it will home the machine to a position of (0, 0, 0.5) - as you have position_endstop=0.5 in your config file.  This is a good setting to use, as you don't want to actually contact the bed during homing.  So, as near as I can tell, your setup is working as intended.  You've configured the machine to descend slightly below the nominal z endstop trigger position when running BED_SCREWS_ADJUST, and it looks like the code is doing just that.

-Kevin

marcosscriven, 2020-05-06:
Thanks for the explanation @KevinOConnor - sorry for the confusion.

closed 2020-05-06 by marcosscriven