Klipper3d/klipper #2759: Recv: !! bed_mesh: Invalid y-axis table length

gshorten, 2020-04-22:
Bed mesh is failing on my Delta, I get this error message:
Recv: !! bed_mesh: Invalid y-axis table length
It used to work before,  I believe these failures are occuring after I did a git pull last week.  Everything else works well.
The failure occurs when the probe is  near the y tower, probably the 2nd or 3rd last probe. I have tried different mesh_radius, mesh_origin, round_probe_count settings, same result always.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4513421/klippy.log)

Arksine, 2020-04-22:
It isn't valid to add a "points" option to bed_mesh.  Explicit checks aren't done against it, but its resulting in a conflict.  The module must generate its points to guarantee that they are valid.  I would recommend increasing the round_probe_count to at least 5, you won't generate much of a mesh on a delta with it set to 3 (as you can see, it only probe's 5 points on the whole bed).  

I will add config checks in a future patch to make sure that both of these issues are taken care of, in the meantime you should update your configuration.  You may not want to change the mesh_origin, if you leave it to the default the points will be generated in such a way that the bltouch probes 0,0.

gshorten, 2020-04-22:
Thanks for the help! I can see what happened now... I was wondering where the "points" setting came from, as I did not put one in the bed_mesh section of printer.cfg.  It happened because I  have also tried out the bed_tilt feature - when I went back to bed_mesh I commented out bed_tilt but did not comment out the points line in the bed_tilt section, it ended up getting appended to the bed_mesh section.   As soon as I looked at the log file (as opposed to my printer.cfg file with all its comments) the error jumped right out at me.  It works fine now, except like others I am seeing a high rate of BLTouch "BLTouch failed to deploy" errors (I've been through all the other related tickets but that's something for another day).

The docs are explicit that bed tilt and bed mesh are incompatible, I assumed then that you should also then not  have both sections in a printer.cfg file.
  
BTW I only had round_probe_count set to 3  for debugging, Previously I was using 5 or 7 points.   re, mesjh_origin, I found the doc for this to be confusing, ie is it probing where the nozzle would be (with correct offsets in BLTouch section), if so, why have mesh_origin section?  Is it so you can do a large area probe without parts of the head hitting a tower? But then the mesh is offset?

Thanks again for your help.  As often the case it was something stupid I did :-(

Arksine, 2020-04-22:
You are on the right track with the `mesh_origin` option.  When you have a probe it may be useful to change the origin of the mesh to allow for a larger mesh_radius.  The bed_mesh module does adjust for probe location, so it will indeed move the probe over the mesh_origin during calibration, not the nozzle.  

Adjusting the mesh_origin does move the mesh so it would not be centered on the bed.  Whether or not this is desirable someone depends on the bed and how far the probe is offset from the nozzle on X and/or Y.

closed 2020-04-23 by gshorten