Klipper3d/klipper #2760: [Errno 13] Permission denied: '/dev/serial/by-id/usb-1a86_USB2.0-Se rial-if00-port0'

Hairyloon, 2020-04-22:
I ran out of space so I decided to go for a bigger SD card and a fresh install. 
I thought also I would test my theory about needing to be installed as pi because I didn't notice it say so in the instructions, except insofar as they assume to be starting on a new system.

It hasn't worked, and it did after I cleared out and went back in as pi.so it looks like the theory was right. I've cleaned it all out and gone back in as pi; is working fine now.

May I suggest that we either find why it didn't work and fix it, or put a bit more emphasis in the instructions on the need for pi?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4513294/klippy.log)

BlackStump, 2020-04-22:
I have klipper installed on a Beaglebone Black as user "debian" and
on a Orangepi Zero+ as user "opi"
It depends on what install script you are using, The default instructions clearly state for a 
"Raspberry Pi"
but if you look at the FAQ it explains what is required
https://www.klipper3d.org/FAQ.html#can-i-run-klipper-on-something-other-than-a-raspberry-pi-3

Hairyloon, 2020-04-24:
Sorry, I wasn't clear. I was talking about installing on a pi, and I'd found that if I installed as a user other than pi, then it didn't work.

KevinOConnor, 2020-05-02:
As indicated in the FAQ, you are welcome to setup Klipper in a custom installation, but then you'll need to figure out all the necessary system settings to make that work.

-Kevin

Hairyloon, 2020-05-02:
Who are you talking to there?
I'm not talking about a custom installation, only about installing it on a pi, as a user that isn't pi.

closed 2020-05-11 by KevinOConnor

Hairyloon, 2020-05-11:
In closing this issue rather than addressing the concern are you basically saying that we should log in as user pi in order to use Klipper?