Klipper3d/klipper #2761: Not an issue - more a request for help on 2-in-1-out extruders

dushyantahuja, 2020-04-22:
I've recently bought a 2-in-1-out nozzle https://www.aliexpress.com/item/4000502710296.html?spm=a2g0s.9042311.0.0.27424c4dv90Qz3 - basically has a Y-splitter built into it.  I've configured it using the following: 
```
[gcode_macro T1]
gcode:
 {% if printer.extruder.temperature > 170 %}
  {% if printer.toolhead.extruder != "extruder1" %}
   SAVE_GCODE_STATE NAME=T1
   M83
   G1 E-5.0 F3600        ; Retract filament as soon as possible...
   ;G91            ; Relative
   ;G1 Z2 F6000        ; ...Then lower the bed...
   ;G90            ; Absolute
   G1 X1 Y150 F18000    ; ...Then get the hell away from the print
   G4 P2000        ; Pause two seconds to cool the retracted filament
   G1 E4.9    F3600    ; Pack the filament to eliminate the tail
   G1 E-50 F6000    ; Yank the filament past the Y adapter and into its Bowden tube
   ACTIVATE_EXTRUDER extruder=extruder1
   G1 E50 F6000        ; Extrude 50mm @ 100mm/s
   G1 E5   F1500        ; Extrude   5mm @  25mm/s
   RESTORE_GCODE_STATE NAME=T1
   M117 Extruder 1 Ready
   UPDATE_DELAYED_GCODE ID=clear_display DURATION=10
  {% endif %}
 {% else %} 
  M117 Heat extruder
 {% endif %}

[gcode_macro T0]
gcode:
 {% if printer.extruder.temperature > 170 %}
  {% if printer.toolhead.extruder != "extruder" %}
   SAVE_GCODE_STATE NAME=T0
   M83
   G1 E-5.0 F3600        ; Retract filament as soon as possible...
   ;G91            ; Relative
   ;G1 Z2 F6000        ; ...Then lower the bed...
   ;G90            ; Absolute
   G1 X1 Y150 F18000    ; ...Then get the hell away from the print
   G4 P2000        ; Pause two seconds to cool the retracted filament
   G1 E4.9    F3600    ; Pack the filament to eliminate the tail
   G1 E-50 F6000    ; Yank the filament past the Y adapter and into its Bowden tube
   ACTIVATE_EXTRUDER extruder=extruder
   G1 E50 F6000        ; Extrude 50mm @ 100mm/s
   G1 E5   F1500        ; Extrude   5mm @  25mm/s
   RESTORE_GCODE_STATE NAME=T0
   M117 Extruder 0 Ready
   UPDATE_DELAYED_GCODE ID=clear_display DURATION=10
  {% endif %}
 {% else %} 
  M117 Heat extruder
 {% endif %}
```
While this works fine, I'm concerned that I might have the T1 filament loaded when the printer starts. Unfortunately, klipper assumes that T0 is activated when it restarts. Now sending the T0 command will not load the filament, and in all probability will jam the filament from T0 into the splitter. Is there any way to save the extruder number across restarts?

KevinOConnor, 2020-04-24:
There is currently no way to store state like that across restarts.  I'd guess it would be possible to introduce a hardware filament sensor switch on both paths and update the macros to check that state for a conflict.

-Kevin

dushyantahuja, 2020-04-24:
Thanks - is there a way for Macros to save the state in the configuration file - similar to what happens when you send SAVE_CONFIG

KevinOConnor, 2020-04-24:
No, there is currently no way to save arbitrary state like that across restarts.

-Kevin

dushyantahuja, 2020-05-01:
I've started creating an extras module to save variables and was wondering if there was any way to run SAVE_CONFIG without issuing a restart from within the module. Initial code here:
https://github.com/dushyantahuja/klipper/blob/master/klippy/extras/save_variables.py

Alternatively, do you think it would be better to create a new file altogether, so that the config file does not get corrupted?

dushyantahuja, 2020-05-04:
Closing this issue, creating a new one for saving variables.

closed 2020-05-04 by dushyantahuja