Klipper3d/klipper #2762: TMC2130's Only Working After Flashing back to Marlin and then back to Klipper

jlwainwright, 2020-04-22:
Hi All

I have not been able to get my TMC2130 steppers running since trying to move over to Klipper.

After many hours I just had to make sure that I am not dealing with a hardware/wiring issue so flashed back to Marlin. Everything worked fine again on Marlin. 

I then flashed back to Klipper to continue troubleshooting (without powering off the printer/pi).

And suddenly everything was working 100% in Klipper as well. I have switched off the printer and on next power-up the TMC's were not working again.

Reading through other issues I wonder if this could not be that the drivers are not powering up on boot-up and that when I flashed Marlin, they got powered and that is why Klipper detected them on the next flash?

Hope the log and DUMP_TMC can shed more light on this:

[klippy.log](https://github.com/KevinOConnor/klipper/files/4514762/klippy.log)

[DUMP_TMC.txt](https://github.com/KevinOConnor/klipper/files/4514764/DUMP_TMC.txt)

BlackStump, 2020-04-22:
Have you issued a RESTART command?
https://www.klipper3d.org/G-Codes.html#tmc2130-tmc2660-tmc2208-tmc2209-and-tmc5160
>INIT_TMC STEPPER=<name>: This command will intitialize the TMC registers. Needed to re-enable >the driver if power to the chip is turned off then back on.

BlackStump, 2020-04-22:
This might be worth having a read of
https://github.com/KevinOConnor/klipper/issues/1702

KevinOConnor, 2020-04-22:
The log indicates that Klipper is not successfully communicating with the tmc drivers.  Double check the cs pin definitions, and it may be necessary to specify the spi pins (via software spi).  See the config/example-extras.cfg file for further details.

-Kevin

closed 2020-05-11 by KevinOConnor