Klipper3d/klipper #2763: SAVE_CONFIG does not save M206's values

karabas2011, 2020-04-22:
Big delta. Mks Sbase+Raspberry/octoprint.
In order to fast adjusting 1st layer I used to use M206 Zxxx (or SET_GCODE _OFFSET in klipper) and save it with M500
Currently it seems impossible. 
SAVE_GCODE _STATE saves only during session. 
Another way - to change and save z-probe offset is impossible too without the whole calibration procedure.
Am I missed something?

[klipper.log](https://github.com/KevinOConnor/klipper/files/4515203/klipper.log)

BlackStump, 2020-04-22:
M206 is not a supported command neither is M500 afaik
https://www.klipper3d.org/G-Codes.html

You can add a macro with SET_GCODE_OFFSET and/or add that to your start gcode
https://github.com/KevinOConnor/klipper/blob/master/config/sample-macros.cfg

karabas2011, 2020-04-22:
I know. I use SET_GCODE _OFFSET and SAVE_CONFIG but it does not save.

BlackStump, 2020-04-22:
I do not think set_gcode-offset is a save_config item, you would need to manually save it in your config.
>SAVE_CONFIG: This command will overwrite the main printer config file and restart the host >software. This command is used in conjunction with other calibration commands to store the >results of calibration tests.

Hywelmartin, 2020-04-22:
https://github.com/KevinOConnor/klipper/blob/master/docs/Delta_Calibrate.md

KevinOConnor, 2020-05-11:
I'm going to close this as it looks like the original question was answered (SAVE_CONFIG is not intended to save the state of SET_GCODE_OFFSET - edit the config manually for that).

-Kevin

closed 2020-05-11 by KevinOConnor