Klipper3d/klipper #2764: Klipper on TH3D EZBoard

tychop, 2020-04-22:
Does anybody know what options I need to choose during the 'make menuconfig' step of the Klipper installation?

jcritchjr, 2020-04-22:
I am also curious if the TH3D EZBoard Lite V1.2 3d printer controller board can be used with the Klipper software?

Can Klipper generate firmware that will work with the EZBoard?

Where would I find information about configuring the EZBoard for use with Klipper?

I've searched the Github Issues and other sections and have not found my answers.  I've searched the Internet and not found answers.

Where would be the best place to look for this answer or any Klipper supported boards / hardware requirements for the 3d printer portion of Klipper?

Sorry, I don't have a log file because I'm still in the research phase for my use of Klipper.

Thank you!

BlackStump, 2020-04-22:
It seems to be a closed source board from my very quick search, no info on mcu, pins, schematic etc, if you have the board you would need to provide those details, if you can identify the mcu then maybe it is already supported or a good basis to start from

tychop, 2020-04-22:
> It seems to be a closed source board from my very quick search, no info on mcu, pins, schematic etc, if you have the board you would need to provide those details, if you can identify the mcu then maybe it is already supported or a good basis to start from

When I compile marlin for it I use the following command:

	platformio run -e LPC1769 -v && cp .pioenvs/LPC1769/firmware.bin

 Maybe that helps?

Hywelmartin, 2020-04-22:
Don't try to reverse engineer... =)

have a look in the THC ez Marlin clone for pins and other stuff

> SECTION 10 – COUNTERFEITING & CLONING
>
> By purchasing our custom products (EZABL, EZOut, EZBoard, and any products made in house by TH3D) you agree to not reverse engineer, duplicate, copy, or do anything that would duplicate what we are selling here.
>
> If you purchase a product with the intent to copy it for your own gain or someone else’s gain we will cancel your order and deny sales to you. If you get the item and then duplicate it we reserve the right to pursue any and all forms of legal action against you for violating these terms of service.

tychop, 2020-04-23:
There's nothing to reverse engineer. All the pins are in their public fork of Marlin: https://github.com/houseofbugs/TH3D-Unified-U2.R1/blob/master/Marlin/src/pins/lpc1769/pins_TH3D_EZBOARD.h

davidja92, 2020-04-24:
Well you would make config for the lpc1679. From there u would use the sample config and start grabbing the pins from the github th3d posted

tychop, 2020-04-24:
So it's basically creating two configs and run the make again.... I'll give it a go, Thnx

davidja92, 2020-04-24:
Make config change from the atmega and use lpc1769 the exit the menu send the make command then follow rest of install

davidja92, 2020-04-24:
After u can log into octoprint and into the klipper U need to edit your printer.cfg File with all the pins that the3d has posted

Paulbsh, 2020-04-25:
I know the developer of EZBoard pretty well. 
If you need information about it, let me know what exact information you need

KevinOConnor, 2020-05-11:
Any further updates on this?

-Kevin

closed 2020-05-24 by KevinOConnor

th3fallen, 2020-10-13:
@tychop did you ever get this running? if so could you share your config?

TwoPuncakes, 2021-06-03:
@RedlineGT and @tychop, I managed to get Klipper working on my EZBoard/Ender 3 Pro. I haven't printed stuff quite yet though, but my motors, fans, heaters, and LCD seem to functioning perfectly so far. 

So my config is based off of this [config](https://github.com/th3fallen/klipper/blob/feature/add-ezboard-config/config/generic-th3d-ezboard-lite.cfg) by @th3fallen. The important changes I made to his config are:
- Fixed 2 fan pinouts
- Implemented StealthChop
- Added LCD functionality

I can upload an example config, but I want to do a test print first before I do that. Also, I'm not sure how to upload my config to the master repo (I think that's what it's called)?

RedlineGT, 2021-06-03:
This is great news! Post the config file here to get things rolling I suppose. Please do report back. This is exciting!

TwoPuncakes, 2021-06-06:
Done! Here's an example config file: [generic-th3d-ezboard-lite-v1.2.zip](https://github.com/KevinOConnor/klipper/files/6603719/generic-th3d-ezboard-lite-v1.2.zip)
I've included most of the pins from the [pinout map](https://github.com/th3dstudio/UnifiedFirmware/blob/2.0.x/Firmware/Marlin/src/pins/lpc1769/pins_TH3D_EZBOARD.h), so they should be working. HOWEVER, there are some pins in this config file I need to mention:
- EZABL compatibility -  it will probably work with BLTouch pins
- Filament sensor - I don't have nor use one, so I'm not sure if the pins are configured correctly 
- Expansion Header - this is apparently for the "E3D PT100 Temperature Sensor w/ Amplifier Board". I don't have nor use one, so I'm not sure if that will work.
- On-board SD card reader - I doubt anyone will ever use the SD card reader now

And some other notes:
- Most of the printer settings is based off the Ender 3 Pro config, so adjust accordingly with whatever printer you have.
- I used the SKR Mini E3 v2.0 config as a layout template, but I assure you, you won't find those pins here.  
- The run_current for the motors are from TH3D's [configuration_adv.h](https://github.com/th3dstudio/UnifiedFirmware/blob/2.0.x/Firmware/Marlin/Configuration_adv.h). Make sure to enter them correctly in your printer.cfg (600 in Marlin config = 0.600 in Klipper) 
- I am not responsible for potentially destroying your board.

Finally, [here's a test print](https://youtu.be/ZImplD6fM_A)! So far so good!