Klipper3d/klipper #2768: Failed to verify BLTouch probe is raised; retrying.

sugar012, 2020-04-23:
I got this message 90% of the time. Normally t takes 4/5 firmware_restart to get the machine ready. When is conncted it fly like a bird but even with G29 it fails 3/4 probes. 
I searched in the issue but i dont think someone else got my exact error. Needless to say that i tried with marlin and i got no problems. What could it be? Thanks group
[klippy.log](https://github.com/KevinOConnor/klipper/files/4520888/klippy.log)

sugar012, 2020-04-23:
and many times this:
[image]

KevinOConnor, 2020-05-02:
It looks like Klipper failed to get a good clock synchronization upon first connecting to the micro-controller.  This resulted in incorrect scheduling of the initial bl-touch pin_up command, which caused the micro-controller to raise an error.  This would only be an issue upon connecting, because the clock synchronization code would properly sync up with the board after a few seconds.

I updated the Klipper code to try and improve the initial clock synchronization (commit f4429a54).  I recommend you update to the latest code (see the FAQ for details).

-Kevin

KevinOConnor, 2020-05-02:
I should also mention that the "Failed to verify BLTouch probe is raised; retrying" is unrelated to the "timer too close" error I described above.  See https://www.klipper3d.org/BLTouch.html for information on the bltouch error.

-Kevin

sugar012, 2020-05-02:
Thanks Kevin. i updated (both klipper and firmware) and also the issue with bltouch seems resolved. I saw you worked on that too. 
But i tried restarting only the printer but it need a firmware_restart to connect. Is it working as intended or still something wrong? Klippy.log attached

Thanks
[klippy.log](https://github.com/KevinOConnor/klipper/files/4568237/klippy.log)

KevinOConnor, 2020-05-02:
The log you attached shows the old version of the code.  If the problem persists, try attaching a new log (running the latest code).

-Kevin

sugar012, 2020-05-02:
here, sorry.
The BLtouch error returned
[klippy.log](https://github.com/KevinOConnor/klipper/files/4568368/klippy.log)

artheoday, 2020-05-02:
I have this same error... 
[image]

KevinOConnor, 2020-05-11:
@sugar012 - all of your logs still show the old version of Klipper.  If you're still looking for assistance, I suggest performing a clean checkout of Klipper and retry.  If you do post a log, please triple check it is from the latest code (the log will contain the version information).

-Kevin

closed 2020-05-24 by KevinOConnor