Klipper3d/klipper #2772: SKR 1.4 with TMC2209 and BLTouch

Xenomorphdelombre, 2020-04-24:
Hi,  swtched from the Ender 3 board to SKR 1.4 with TMC2209 and BLTouch.

I managed to get the printer back on, xyz move but the BLTouch does not work and I don't have temp nor can I heat the bed or the extruder.

Attached is Klippy.log

[klippy.log](https://github.com/KevinOConnor/klipper/files/4526094/klippy.log)

davidja92, 2020-04-24:
Sounds like a config issue post up your config

Xenomorphdelombre, 2020-04-24:
> # This file contains common pin mappings for the BIGTREETECH SKR V1.4
# board. To use this config, the firmware should be compiled for the
# LPC1768.

# See the example.cfg file for a description of available parameters.<

[stepper_x]
step_pin: P2.2
dir_pin: !P2.6
enable_pin: !P2.1
step_distance: .0125
endstop_pin: P1.29
position_min: 3
position_endstop: 3
position_max: 255
homing_speed: 100

[tmc2208 stepper_x]
uart_pin: P1.10
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

[stepper_y]
step_pin: P0.19
dir_pin: !P0.20
enable_pin: !P2.8
step_distance: .0125
endstop_pin: P1.28
position_min: 0
position_endstop: 0
position_max: 235
homing_speed: 100

[tmc2208 stepper_y]
uart_pin: P1.9
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

[stepper_z]
step_pin: P0.22
dir_pin: P2.11
enable_pin: !P0.21
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_min: -5
position_max: 250

[tmc2208 stepper_z]
uart_pin: P1.8
microsteps: 16
run_current: 0.650
hold_current: 0.450
stealthchop_threshold: 30

# This is the part cooling fan that plugs right in to the fan output marked 'FAN'
[fan]
pin: P2.3

# Ender 3 stock heater fan wires into the second extruder heater output marked 'HE1'
[heater_fan e0_fan]
pin: P2.4
heater: extruder
heater_temp: 50.0
fan_speed: 1.0

# Ender 3 stock heater bed
[... 292 lines cut ...]
name = "  -0.1"
width = 7
gcode = TESTZ Z=-0.1

[menu __calibration_probe_calibrate_testz_plus_point_1]
cursor = \x20
type = command
name = "  +0.1"
width = 7
gcode = TESTZ Z=+0.1

[menu __calibration_bed_mesh_calibrate]
type = command
name = Generate bed mesh
width = 18
show_back = False
gcode = 
	G28
	BED_MESH_CALIBRATE
	SAVE_CONFIG+

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
<

davidja92, 2020-04-24:
Config looks right for the 1.4 I would change  your tmc sections to 2209 since you are using that...if u activate temps do u feel them warming up even if they report 0?

Xenomorphdelombre, 2020-04-24:
I wil lchange the names for TMC2209, Set heat and type status?

When I try to heat up nothing happens

davidja92, 2020-04-24:
Did u ensure your thermistors are plugged in properly into you board?

Xenomorphdelombre, 2020-04-24:
Yes I even tried other thermistor. Also plugged back old board, everything works. Status is Ready. I can move all axis

davidja92, 2020-04-24:
Did you fix header to tmc2209 save it and restart firmware?

Xenomorphdelombre, 2020-04-24:
I tried to go back to Marlin and I got a MAXTEMP error on boot.  Board is labeled as BigTreetech SKR 1.4 ARM Smoothieboard
<blockquote class="imgur-embed-pub" lang="en" data-id="a/EZvnCHE"><a href="//imgur.com/a/EZvnCHE"></a></blockquote><script async src="//s.imgur.com/min/embed.js" charset="utf-8"></script>

davidja92, 2020-04-24:
I saw I klipper log you attempting to set temp but your temp reading reports 0

Xenomorphdelombre, 2020-04-24:
Power in DC IN, Bed in HB, Hot end on HE0 and Hotend fan on HE1. Thermistor are in: Bed: TB and hotend on TH0

Xenomorphdelombre, 2020-04-24:
Also BLtouch does a self test when power on but doesn't deploy when homing

davidja92, 2020-04-24:
Are you 1.4 or 1.4turbo?

Xenomorphdelombre, 2020-04-24:
I believe it is a 1.4 the processor is  LPC1768

davidja92, 2020-04-24:
Ok so visual of your board shows the nxp lpc1768 that would be non turbo

davidja92, 2020-04-24:
Do you have your power jumper on vdd not usb?

Xenomorphdelombre, 2020-04-24:
Yes VDD

davidja92, 2020-04-24:
Sorry just to verify I have the red wires on the left and black on the right to power your bed and extruder?

davidja92, 2020-04-24:
And check your 2 fuses

Xenomorphdelombre, 2020-04-24:
> Sorry just to verify I have the red wires on the left and black on the right to power your bed and extruder?

Yes, Check both fuses, they are good 

This ia an odd one eh? :(

davidja92, 2020-04-24:
Yeah you have exact same setup as me

davidja92, 2020-04-24:
Does your bed mossfet power up? Should have an always on led and when u send power it should turn on another

Xenomorphdelombre, 2020-04-24:
the always on Red is there, but when I send heat nothing lights up. As if the pins are wrong

davidja92, 2020-04-24:
Do you have a spare mossfet to test ? Make sure it didn’t fail

davidja92, 2020-04-24:
Hopefully it isn’t a bad board :( try hitting the reset button on the board itself

davidja92, 2020-04-24:
When you went back to marlin the max temp error is cause by thermsistors broken it not plugged in

davidja92, 2020-04-24:
Also why you bed heat shows 0.... verify wires are not broken and you don’t have them screwed down too tight

Xenomorphdelombre, 2020-04-24:
I don't have another mosfet. The thermistor still works fine on the original Creality board. 

I haven'T try hitting reset... Let's see!

Xenomorphdelombre, 2020-04-24:
Reset doesn't do anything. 

I tried Marlin without the probe connected as well as with the Define disable bogus maxtemp error. I was getting the same error. It start to look like a bad board...

davidja92, 2020-04-24:
Join the klipper discord and post on the skr channel so we can get more people on this

Xenomorphdelombre, 2020-04-24:
can you give me the link?

davidja92, 2020-04-24:
https://discordapp.com/invite/df9yTE2

Xenomorphdelombre, 2020-04-24:
Oh when I try to heat the hotend, there is one of the mosfet light that light up and then looks like the printer crashed. I couldn't control anymore had to power cycle

davidja92, 2020-04-24:
You have an external mossfet on your hotend? External mossfet normally only for heatbed

davidja92, 2020-04-24:
When you lose connection go into the terminal of octopi to see what the errors are

Xenomorphdelombre, 2020-04-24:
No external mosfet.

Neverming I understand what happened. This is the HE1 fan that started because of
'HE1'
[heater_fan e0_fan]
pin: P2.4
heater: extruder
heater_temp: 50.0
fan_speed: 1.0

davidja92, 2020-04-24:
Ahh ok

davidja92, 2020-04-24:
Can you post up some nice pics of your board and how everything is plugged in so I can just comb through it

Xenomorphdelombre, 2020-04-24:
https://imgur.com/a/L2chA5G

davidja92, 2020-04-24:
Everything looks right I don’t get it...make sure the screws for the power wires all the way right and the actual metal wires making contact?

davidja92, 2020-04-24:
Does the terminal report anything when u try to send heat to hot end or bed?

Xenomorphdelombre, 2020-04-24:
I've got a bunch of these.

```
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
```

davidja92, 2020-04-24:
Check the 3 boxes to suppress all those messages

davidja92, 2020-04-24:
And send a m140 s50

davidja92, 2020-04-24:
And advise on what the terminal returns

Xenomorphdelombre, 2020-04-24:
Send: M140 S50
Recv: ok

davidja92, 2020-04-24:
Put your hand on the bed is it getting warm?

Xenomorphdelombre, 2020-04-24:
It is not, and the mosfet lef doesn't light up

davidja92, 2020-04-24:
Klipper will error out after a 20 second timeout since it doesn’t have thermsistor showing the bed heating so just try to see if u feel any heat before the 20 or so seconds

davidja92, 2020-04-24:
Verify your mossfet connection

davidja92, 2020-04-24:
Any heat to your extruder by chance?

Xenomorphdelombre, 2020-04-24:
No both are equialy cold. I tried to heat the bed and checked with a multimeter and no voltage on either bed or hotend screws

davidja92, 2020-04-24:
You have a 3 wire or 4 wire mossfet?

Xenomorphdelombre, 2020-04-24:
I'm not sure. See picture of the mosfet here: https://imgur.com/a/vhDvJUY

davidja92, 2020-04-24:
On your hot bed can you refer the wires? And try to warm it up again

Xenomorphdelombre, 2020-04-24:
refer?

davidja92, 2020-04-24:
Reverse sorry..... online shows the board has the + and - backwards and it shouldn’t matter since the bed has no polarity since it’s just a resistor but some say switching it has worked

Xenomorphdelombre, 2020-04-24:
Switched it, same result. I even switched one of the thermistor. Same result

davidja92, 2020-04-24:
On you printer.cfg try changing your hotbed sensor pin to .23 instead of .25 let me know what happens

davidja92, 2020-04-24:
If nothing move your hotbed thermsistor to th1 with the sensor pin still .23 and if nothing go back to normal...starting to sound like a bad bird out of ideas

Xenomorphdelombre, 2020-04-24:
yeah I tried that. Looks like there is a portion of the board that is dead. Not all of it but maybe something was missed in manufacture. I will try to get it exchanged

Thanks for your help!

[2 more replies]

closed 2020-05-02 by Xenomorphdelombre