Klipper3d/klipper #2773: After a print has started, nozzle fluctuates around target temp and takes forever to start

Daitaro302, 2020-04-24:
Hello all!

I'm noticing something that I would honestly like to see act differently. After I start a print, the printer heats up, like normal. However, once it reaches the temp, it seriously just hovers around it, both above and below. If I'm targeting 220 for instance, it will hit 220, go up to 225, down to 218, back up to 227, and literally just float around the target for up to two minutes or so before starting the print. This isn't new; it has done this since I changed over to klipper. I'm running a cr-10s, with modifications to the printer itself - none related here. This printer prints *beautifully* , this problem *only* happens during warm-up. I don't have an issue with it really, but it's very inconvenient having to sit there and clean the nozzle each print. Seriously, it hovers around for so long that by the time the print starts, there's a dime-sized blob hanging around the nozzle.

Are there some better targeting options I could try out in the config files? I don't have any fans blowing around the printer, the humidity is low; this isn't a crazy environment here. Very normal. 

Thanks, guys!

Hywelmartin, 2020-04-24:
if a PID tune doesn't help.. try lower the _max_power: 1.0_ to the hotend and/or _pid_integral_max:_
lowering _smooth_time: 2.0_ might help

pusewicz, 2020-04-27:
@Daitaro302 Please make sure you calibrate the PID:

https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md#calibrate-pid-settings

closed 2020-05-01 by klipper-gitissuebot