Klipper3d/klipper #2774: LED Light control

MarsArtis, 2020-04-24:
First of all Forgive for my noobness.
Lets' say that till now I've used My Delta with a MK4firmware.
The led is connected to the pin n.7 and I was fine controlling it using M42 gcode, so:

- M42 P7 S0 >LED OFF
- M42 P7 S255 > LED MAX light
- M42 P7 S1 > LED min light

now, I've set my printer.cfg with these lines:
```
[output_pin LED_pin]
pin: ar7
value: 0
```
so to have the Light off on startup.
then added 3 gcodes macros:
```
[gcode_macro LEDOFF]
gcode:  SET_PIN PIN=LED_pin VALUE=0

[gcode_macro LEDMIN]
gcode:  SET_PIN PIN=LED_pin VALUE=0.5

[gcode_macro LEDMAX]
gcode:  SET_PIN PIN=LED_pin VALUE=1
```
LEDOFF and LEDMAX work ok, but LEDMIN returns the following Unhandled comunication error:
**ERROR:invalid pin value**
Why?

Hywelmartin, 2020-04-24:
Have a look here
https://github.com/KevinOConnor/klipper/blob/43fa41c1af648ea626040e17f0eebda39dcdb2cb/config/example-extras.cfg#L1224

MarsArtis, 2020-04-24:
here it comes the log, My bad not attacching.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4529162/klippy.log)

> Have a look here
> https://github.com/KevinOConnor/klipper/blob/43fa41c1af648ea626040e17f0eebda39dcdb2cb/config/example-extras.cfg#L1224

@Hywelmartin 
I set my printer.cfg just by reading that. But maybe I missed something.

MarsArtis, 2020-04-24:
So Following @Hywelmartin tip I modified the .cfg this way:

```
[output_pin LED_pin]
pin: ar7
pwm: True
value: 0

[gcode_macro LEDOFF]
gcode:  SET_PIN PIN=LED_pin VALUE=0

[gcode_macro LEDMIN]
gcode:  SET_PIN PIN=LED_pin VALUE=.5

[gcode_macro LEDMAX]
gcode:  SET_PIN PIN=LED_pin VALUE=1
```
now, when I issue LEDMIN command the lights blink instead of beeing steady and dimmed at 50% of the full power.

[klippy.log2.zip](https://github.com/KevinOConnor/klipper/files/4529294/klippy.log2.zip)

Hywelmartin, 2020-04-24:
Try to use a faster PWM..
[output_pin LED_pin]
pin: ar7
pwm: True
value: 0
cycle_time: 0.010

```
#cycle_time: 0.100
#   The amount of time (in seconds) per PWM cycle. It is recommended
#   this be 10 milliseconds or greater when using software based PWM.
#   The default is 0.100 seconds for pwm pins.
```

or use hardware PWM
hardware_pwm: True

```
#hardware_pwm: False
#   Enable this to use hardware PWM instead of software PWM. When
#   using hardware PWM the actual cycle time is constrained by the
#   implementation and may be significantly different than the
#   requested cycle_time. The default is False.
```

MarsArtis, 2020-04-24:
> Try to use a faster PWM..
> [output_pin LED_pin]
> pin: ar7
> pwm: True
> value: 0
> cycle_time: 0.010
> 
This one is the one that worked, Thks.

> 
> or use hardware PWM
> hardware_pwm: True
> 
> ```
Tried the second setup too but it gave another **ERROR:invalid pin value** .

KevinOConnor, 2020-05-02:
Okay, lets close this as it looks like a solution was found for the original question.

-Kevin

closed 2020-05-02 by KevinOConnor