Klipper3d/klipper #2777: MKS SBASE error: Serial connection closed

AlexSkarbo, 2020-04-24:
Hello.
I decided to upgrade the klipper from version 0.7 to version 0.8, before upgrading the printer, in the configuration file I changed only the port address. 

[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1768_16F0FF16A10CF6AEB6C5A357C12000F5-if00

And now I constantly get a connection error.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4529767/klippy.log)

AlexSkarbo, 2020-04-24:
After some time, the MKS SBASE and RPi connected, I turned on the bed heating and after a few minutes, the connection was interrupted. Now I see Klipper state: Shutdown and Lost communication with MCU 'mcu'.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4530283/klippy.log)

KevinOConnor, 2020-05-02:
I'm not aware of any regressions with usb connectivity.  The best information we have on the "lost communication with mcu" errors is described in the FAQ.

-Kevin

closed 2020-05-24 by KevinOConnor