Klipper3d/klipper #2778: Verify M112 - Error: Shutdown due to M112 command

MarsArtis, 2020-04-24:
I'm following  the Config_checks.md steps.

Problem related to **VERIFY M112** step:
when M112 is issued the board remains OFF, I'm not able to reconnect to the printer.
I have to go on ssh and issue **sudo service klipper start**.
Is it a normal behaviour?

Following the guide it seems like that the board should.
_I'm using Octoprint on a Rasp3 B
Anycubic Delta with Trigorilla standard board_

[klippy.log3.zip](https://github.com/KevinOConnor/klipper/files/4530563/klippy.log3.zip)
 be back on start mode.

Hywelmartin, 2020-04-24:
try another path to the serial

instead of 
`serial = /dev/ttyUSB0`

use something like 
`/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0`

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

MarsArtis, 2020-04-24:
Maybe it's related to the above issue:
I've tried to make a test print, started and at a certain point of the first layer the printer stopped and octoprint lost connection with the Trigorilla board.
It entered in the unconnectable state as above with the **Lost communication with MCU 'mcu'** message.
So again I had to SSH.
BUT, 
**sudo service klipper start** was not working and had to issue **sudo service klipper stop** and then again **sudo service klipper start** to have the trigorilla display show temps and the other infos.

Anyway, OCTOPRINT is not able to connect and continue to report:
```
 Lost communication with MCU 'mcu'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
Lost communication with MCU 'mcu'
```

So back on SSH to try FLASH again with 
make flash FLASH_DEVICE=/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0

but gettin 
```
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
```

[klippy.log4.zip](https://github.com/KevinOConnor/klipper/files/4531059/klippy.log4.zip)

MarsArtis, 2020-04-24:
> try another path to the serial
> 
> instead of
> `serial = /dev/ttyUSB0`
> 
> use something like
> `/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0`
> 
> https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

you wrote meanwhile I was updating my post.
In m y cfg, I had already set serial accordingly

MarsArtis, 2020-04-24:
I'd like to add that I read [https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-keep-getting-random-lost-communication-with-mcu-errors](url) and have to say that I've been using Octoprint connected via usb for about 2 years without any issue.

KevinOConnor, 2020-05-02:
I'm not sure there's much to add here - the best information we have on "lost communication with mcu" errors is in that FAQ entry.

-Kevin

closed 2020-05-24 by KevinOConnor