Klipper3d/klipper #2780: MCU 'mcu' Shutdown: Timer Too Close

mordhau5, 2020-04-24:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4531552/klippy.log)

Been trying to hunt this one down for a while. Initially, I thought it was related to running timelapses from Octoprint (either Octolapse or built-in), but I have since seen timeouts when the printer is sitting doing nothing. The log attached is from after a print had already finished and was sitting still for a couple minutes. I've even see it fail running klipper macros I've written (in particular the "cold pull" macro, which doesn't work very well btw).

I seem to get it _less_ often when not running timelapses, but it still does happen and I don't have empirical evidence to prove the correlation. I've tried different versions of Octoprint, safe-mode, etc. But ultimately I NEED to be running those timelapses as that's the whole reason I picked up the printer.

Pi3B/Octopi 0.17.0/Octoprint 1.4.0/Wrecklabs PrintHat Mainboard

My understanding from reading other Git Issues is that the error comes when the G-code stream to Klipper falls off for too long, but then why would I get it when the printer is sitting still?

KevinOConnor, 2020-04-29:
It looks like the host software didn't get any processing time for an extended period of time (several seconds) - even though the host software wasn't doing much of anything at the time.  (Ultimately it was the turning off of the heater fan that caused the micro-controller to detect that the communication was not working properly.)  It's certainly an unusual result.

I'd recommend updating to the latest version of the software (see the FAQ for upgrade instructions).  I'm not sure the latest version will resolve the issue, but it does have additional debugging that could help track down an error like this.  There have also been some important stm32f103 fixes since your last update which may be complicating things.

-Kevin

mordhau5, 2020-04-29:
Thanks, @KevinOConnor. I wonder if rendering a timelapse caused the system to lock up in that case? But I know I'm only grabbing 3 of my 4 available threads to render the timelapse. I'll try dropping total threads to 2 instead of 3. But I know there are plenty of times when the CPU should be doing nothing else (and Top confirms this) and it still times out.

Here I have another case where right in the middle of heating the bed and nozzle, the MCU timed out. I know for a fact this time a timelapse was not running or rendering.

[klippy (2).log](https://github.com/KevinOConnor/klipper/files/4551978/klippy.2.log)

Meanwhile, I'll get that software updated and do some more testing.

mordhau5, 2020-05-16:
Haven't gotten this one in a long while. Going to assume it was power spikes/dips from an unreliable mod to an ATX PSU causing communication issues. Switched to a Meanwell and updated to newest commit (at the time) saw this issue disappear.

closed 2020-05-16 by mordhau5