Klipper3d/klipper #2784: Cant FLASH Firmware after upgrade Klipper

StefanMeGit, 2020-04-25:
Sorry guys to bother u with this easy problem, but I cant flash my Arduino for some reason.

Has anyone an idea? It worked flawless before.

```
pi@octopi:~/klipper $ ls /dev/serial/by-path/*
/dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0
pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0
  Flashing out/klipper.bin to /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0
Entering bootloader on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0
Device reconnect on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0
lib/bossac/bin/bossac -U -p /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0 -e -b -w out/klipper.bin -v

No device found on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0
Failed to flash to /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0: Error running bossac
src/atsam/Makefile:53: recipe for target 'flash' failed
make: *** [flash] Error 255
```

StefanMeGit, 2020-04-25:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4533131/klippy.log)

Sorry, I thought it doesn't matter because flashing is not part of this.

Hairyloon, 2020-04-27:
It's an automated script: it asks for a log automatically.
I can only suggest the obvious: had you disconnected octopi and shut down klipper?
Was there anything else running that might have captured the serial port?

KevinOConnor, 2020-05-11:
The attached log shows Klipper working fine - so it doesn't look like there's anything we can assist with.

-Kevin

closed 2020-05-24 by KevinOConnor