Klipper3d/klipper #2785: Does it have to flounce off every time?

Hairyloon, 2020-04-25:
Is it absolutely necessary for klipper to disconnect and sulk after every little thing that goes wrong?
I get that you don't want it to move if it hasn't homed (except sometimes, when it's buried the hotend in the workpiece), but does it have to disconnect as well as refuse?
And I'm not sure that it is supposed to, but mine requires a firmware restart almost every time.

I wouldn't mind so much if it noticeably protected the thing from doing stupid things, but I've lost count of the number of times that it's tried to ram the head through the bed.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4533161/klippy.log)

Coffee0297, 2020-04-25:
klipper don't disconnedct octo do. you have not setup after the manual look here: https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#configuring-octoprint-to-use-klipper

its a setting you forgot

kind regards Tonny

Hairyloon, 2020-04-25:
> klipper don't disconnedct octo do. 
That's overly pedantic. Klipper tells it to disconnect. And half the time Klipper has to be stopped and restarted before it will connect again.

>you have not setup after the manual look here: https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#configuring-octoprint-to-use-klipper

Yeah, if I hadn't configured octoprint then how come it connects?

Coffee0297, 2020-04-25:
i was refering to this 
``
In the Settings tab, navigate to the "Behavior" sub-tab and select the "Cancel any ongoing prints but stay connected to the printer" option. Click "Save"
``

kind regards Tonny

Hairyloon, 2020-04-26:
Ah, well I'd done that... definitely.
But then I re-installed octopi and yes, I'd forgotten to do that bit.
Thank you.

pusewicz, 2020-04-27:
@Hairyloon Please close the issue if the problem got resolved. Thanks!

Hairyloon, 2020-04-27:
Sorry, I wanted to test it before I closed it, and hadn't had a chance to.

closed 2020-04-27 by Hairyloon