Klipper3d/klipper #2786: tmc5160 SPI registers endian backwards??

ricochet1k, 2020-04-25:
I'm trying to get my TMC5160 motors to work with a RADDS board that I have added the SPI wiring to. And funny story, I had the MISO pin shorted to ground by accident, and when I tried to buzz the stepper *without* the tmc5160 section in the config file, the motor turned fine. 

When I try to add the tmc5160 section (and I fixed the wiring) I get a very brief click from the motor and then nothing. Running the DUMP_TMC command gives strange output (values set in locations not mentioned in the datasheet). Attached is a log of me running the DUMP_TMC command several times in a row. Flags that don't seem like they should change frequently (short to ground, open load, etc.) are changing every time. DRV_STATUS in particular, I've verified with my voltmeter that none of the motor pins are shorted to ground, but the s2g, ol, ot values flicker on and off randomly. From reading the datasheet, bits on the other end of DRV_STATUS are SG_RESULT, which appears to be a measurement of some motor load something or other that I would expect to change somewhat randomly on the low-significance end.
[dump_tmc5160.txt](https://github.com/KevinOConnor/klipper/files/4534278/dump_tmc5160.txt)

Best I can guess is that the bits or the bytes of the registers are being read and possibly written backwards, but I'm clueless as to why that would happen to me and not anybody else.

Also, looking at the results of the DUMP_TMC command in #2047, those bits look more normal.

Any idea what is going on? 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4534292/klippy.log)

KevinOConnor, 2020-04-30:
As a guess, you may be hitting the same issue described at #2086 - specifically some stepsticks exported the "clk" pin and if that's not properly isolated it can cause havoc.

-Kevin

KevinOConnor, 2020-05-11:
I'm going to close this as it looks like the conversation has concluded.  To the best of my knowledge, Klipper works as intended.

-Kevin

closed 2020-05-11 by KevinOConnor