Klipper3d/klipper #2788: Klipper with RADDS board not correct working

bixicy, 2020-04-26:
Hi,

Im new on Klipper Firmeware.

I installed Klipper on Arduino Due original. Im working with Repetier Server and not Octoprint. And set some settings on printer.cfg.

But I cant read the temperatures from both Thermistors. Both say 0 degrees.
And the second Issue is, the Hotend heater is permanent on. But on the Repetier Frontend it says off.

My printer.cfg
```
# This file contains common pin mappings for RADDS (v1.5) boards.  To
# use this config, the firmware should be compiled for the Arduino
# Due.

# See the example.cfg file for a description of available parameters.

# Temp sensor pins: analog0..analog4
# Mosfet Pins: ar7 (Heatbed), ar8, ar9, ar11, ar12, ar13

[stepper_x]
step_pin: ar24
dir_pin: ar23
enable_pin: ar26
step_distance: .0125
endstop_pin: ^ar28
#endstop_pin: ^ar34
position_endstop: 0
position_max: 330
homing_speed: 50

[stepper_y]
step_pin: ar17
dir_pin: !ar16
enable_pin: ar22
step_distance: .0125
endstop_pin: ^ar30
#endstop_pin: ^ar36
position_endstop: 0
position_max: 310
homing_speed: 50

[stepper_z]
step_pin: ar2
dir_pin: ar3
enable_pin: ar15
step_distance: .0025
endstop_pin: ^ar32
#endstop_pin: ^ar38
position_endstop: 0.5
position_max: 400

[extruder]
step_pin: analog7
dir_pin: analog6
enable_pin: analog8
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar13
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog0
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 275

#[extruder1]
#step_pin:           analog10
#dir_pin:            analog9
#enable_pin:         analog11

#[extruder2]
#step_pin:           ar51
#dir_pin:            ar53
#enable_pin:         ar49

[heater_bed]
heater_pin: ar7
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog1
control: watermark
min_temp: 0
max_temp: 150

[verify_heater heater_bed]
# adjust for personal bed setup, this prevents stock heated bed from issuing
# false positive heating errors due to slow temperature increase
check_gain_time: 600

[fan]
pin: ar9

[heater_fan nozzle_cooling_fan]
pin: ar8

[mcu]
serial: /dev/serial/by-id/usb-03eb_6124-if00

[printer]
kinematics: corexy
max_velocity: 300
max_accel: 1000
max_z_velocity: 20
max_z_accel: 100

# "RepRapDiscount 2004 Smart Controller" type displays
#[display]
#lcd_type: hd44780
#rs_pin: ar42
#e_pin: ar43
#d4_pin: ar44
#d5_pin: ar45
#d6_pin: ar46
#d7_pin: ar47
#encoder_pins: ^ar52, ^ar50
#click_pin: ^!ar48

# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
#[display]
#lcd_type: st7920
#cs_pin: ar42
#sclk_pin: ar44
#sid_pin: ar43

```

[klippy.log.txt](https://github.com/KevinOConnor/klipper/files/4536346/klippy.log.txt)

jakep82, 2020-04-27:
What options did you use in the `make menuconfig` before flashing the Due?  The log you posted doesn't show the Due ever connecting.

bixicy, 2020-04-27:
This is from Klipper config file
```
# Automatically generated file; DO NOT EDIT.
# Klipper Firmware Configuration
#
# CONFIG_LOW_LEVEL_OPTIONS is not set
# CONFIG_MACH_AVR is not set
CONFIG_MACH_ATSAM=y
# CONFIG_MACH_ATSAMD is not set
# CONFIG_MACH_LPC176X is not set
# CONFIG_MACH_STM32 is not set
# CONFIG_MACH_PRU is not set
# CONFIG_MACH_LINUX is not set
# CONFIG_MACH_SIMU is not set
CONFIG_STEP_DELAY=2
CONFIG_BOARD_DIRECTORY="atsam"
CONFIG_MCU="sam3x8e"
CONFIG_CLOCK_FREQ=84000000
CONFIG_USBSERIAL=y
CONFIG_ATSAM_SELECT=y
CONFIG_MACH_SAM3X8E=y
# CONFIG_MACH_SAM3X8C is not set
# CONFIG_MACH_SAM4S8C is not set
# CONFIG_MACH_SAM4E8E is not set
CONFIG_MACH_SAM3X=y
CONFIG_FLASH_START=0x80000
CONFIG_FLASH_SIZE=0x80000
CONFIG_RAM_START=0x20000000
CONFIG_RAM_SIZE=0x18000
CONFIG_STACK_SIZE=512

#
# USB ids
#
CONFIG_USB_VENDOR_ID=0x2341
CONFIG_USB_DEVICE_ID=0xabcd
CONFIG_USB_SERIAL_NUMBER_CHIPID=y
CONFIG_USB_SERIAL_NUMBER="12345"
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
```

bixicy, 2020-04-28:
Hi,
i delete ALL and started this time with octopiimage. Formatted SD Card and booted with f´new OctoPi on Raspberry.

After installing and configuration klipper, the same issue as with repetier server.

Cant see temps and issues in serial connection
```
Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 186, in send_with_response
    return src.get_response(cmd, self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 242, in get_response
    self.serial.raw_send_wait_ack(cmd, minclock, minclock, cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 178, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
```
In attachment some new files:
[klippy.log.txt](https://github.com/KevinOConnor/klipper/files/4547424/klippy.log.txt)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4547425/printer.cfg.txt)
[.config.txt](https://github.com/KevinOConnor/klipper/files/4547426/default.config.txt)

jakep82, 2020-04-28:
Have you tried connecting to the other USB port on the Due?  I haven't flashed my Due in a while, but if you use serial, you need to connect to the programming port.  If you choose to use USB in the menuconfig, you need to connect to the native USB port after flashing.  You won't see temperature information until the host (Klippy) sees the Due and connects to it.

bixicy, 2020-04-28:
Yes, I tried on both USB ports. But is no changes

jakep82, 2020-04-28:
Please post a new log file showing this error.

bixicy, 2020-04-29:
Hi,
the new log File.

> fix by recompiling and flashing the firmware
I did this several times yesterday, but the error message remains.

[klippy.log.txt](https://github.com/KevinOConnor/klipper/files/4550594/klippy.log.txt)

jakep82, 2020-04-29:
In the [mcu] section you need to have `pin_map: arduino`

bixicy, 2020-04-29:
Ah, nice to know.

I added, restartet klipper via Terminal (putty), and FIRMWARE_RESTART via Octoprint frontend.

I have my Temps. Now I've moved on. Thanks a lot.

Maybe you should take a note in the documentation or in config examples

closed 2020-04-29 by bixicy

jakep82, 2020-04-29:
The RADDS example config does include this line.