Klipper3d/klipper #2792: Blobs / zits on round models

Maxiba, 2020-04-27:
Hello!

I have a problem with getting blobs / zits [20200427_173909] on my round model [Model.zip].
I tried several times. It seems like it has always the same pattern.

Some unsorted informations:
- Axes/Motors on my Sovol SV01 are calibrated and not vibrating. Most probably not a mechanical problem.
- Calibrated the pressure_advance setting (0.105)
- The microcontroller / connection doesnt seem exhausted [loadgraph3], as i tried printing from virtual SD too.
- I'm using OctoKlipper
- The printer is stuttering. I think the stuttering is causing the blobs.
- Dont have any issues on non-round objects [20200427_173939]

[image]
[image]
[image]
[klippy3.log](https://github.com/KevinOConnor/klipper/files/4540654/klippy3.log)
[Model.zip](https://github.com/KevinOConnor/klipper/files/4540583/Model.zip)

Best Regards!
Max

jalanjarosz, 2020-04-28:
Do you have the “Display Layer Progress” plug-in installed and enabled?   

I only mention this because there is a known issue with this popular plug-in that causes stuttering, pauses, and this blobs.

Maxiba, 2020-04-28:
No, there is only OctoKlipper, BedVisualizer ans Action Command Prompt in my Plugins Settings. I didnt installed any other plugins yet. Or do you mean GCodeViewer?

Coffee0297, 2020-04-28:
what slicer? the new cura 4.6 are know in create issues. 

kind regards Tonny

Maxiba, 2020-04-28:
Yes, actually use Cura 4.6... I just printed a new test cylinder. I will post the result later...
What free slicer can you recommend?

Coffee0297, 2020-04-28:
slic3r++

Hywelmartin, 2020-04-28:
my host buffer is normally way lower than yours.... are you certain that you are printing from vSD

Maxiba, 2020-04-28:
> slic3r++

I tried slicing and printing with slic3r. And it really turned out much better. There arent any blobs or zits! Just a few different printing problems i didnt had with cura, but i think i just need to tune my settings in slic3r. I would have never thought that the problem is my slicer!

> my host buffer is normally way lower than yours.... are you certain that you are printing from vSD

In the graph i have posted i wasnt printing with vSD. But i had the same printing result with and without vSD.

jalanjarosz, 2020-04-28:
There is some good slicer settings here:

https://github.com/KevinOConnor/klipper/blob/master/docs/Slicers.md

And here too:

https://github.com/KevinOConnor/klipper/issues/1869

Thanks!
James

jakep82, 2020-04-28:
Are you sure you were printing from the vSD file?  Just enabling vSD in the printer.cfg is not enough, you also need to make sure you configure Octoprint to only show SD files so you're selecting the correct file to print.  This issue is very common with Octoprint, and I've always been able to solve it with vSD.

Maxiba, 2020-04-29:
> Are you sure you were printing from the vSD file? Just enabling vSD in the printer.cfg is not enough, you also need to make sure you configure Octoprint to only show SD files so you're selecting the correct file to print. This issue is a very common with Octoprint, and I've always been able to solve it with vSD.

Yes, I am sure. The message in terminal said "starting printing from SD" or something like that. And the gcode commands were not showing, it just said, that the commands from SD card are still running. The problem was really cura 4.6. The issue is known and will be fixed this week with release of cura 4.6.1.

See here: https://www.reddit.com/r/klippers/comments/g8z3st/stuttering_with_virtual_sd_too/

closed 2020-04-29 by Maxiba