Klipper3d/klipper #2793: Internal error on command:"G1"

Maarten88, 2020-04-28:
Near the end of a 2.5 days, 1.3kg print, Klipper suddenly stopped. The control display seems hung, showing the printing temperatures but the printer is shutdown and cooled down. OctoPrint is disconnected from Klipper showing an error: "Too many consecutive timeouts".
The log file seems to indicate an internal error in Klipper caused this.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4543561/klippy.log)

This error seems exactly the same as issue #2771. But I run a rather recent version of Klipper, I had updated to the latest version about one week back before starting this large print job, which I now regret as I never had problems before. This print was part 3 of a large part, I did not restart Klipper or OctoPrint in between jobs. Klipper did finish the first two parts (one is a mirror from the one that now failed) without issue, so it had been running and printing nonstop for about a week.

Hywelmartin, 2020-04-28:
got the same
 `Internal error on command:"G1"`
old klipper
reprinted the aborted file 

updated klipper and it is now printing.. i have been printing more than 300 prints on this old klipper install
[klippy_g1.log](https://github.com/KevinOConnor/klipper/files/4544505/klippy_g1.log)

KevinOConnor, 2020-04-29:
@Maarten88 - Was that the full log?  Normally the log contains additional debugging information after the error.

-Kevin

Maarten88, 2020-04-29:
Yes, this was the whole klippy.log file as I got it from the Pi. In the morning I found that the print had stopped early that night. OctoPrint would not reconnect, the control box seemed thoroughly hung, so I turned the control box off, also disconnected USB to cut its power completely to restart it and then also rebooted the Pi from the OctoPrint UI. I think I downloaded this log before the reboot.

Maarten88, 2020-04-29:
I have changed nothing to the setup and simply started the next print job, under the assumption that somehow the long and uninterrupted print jobs caused this. I will now be restarting everything between jobs. The printer has been running for over a day now and has to go for another day before this job finishes.

I saw that there were several changes to the chelper functions that were logged in the stack trace about two month back. Would it be better to go back to a version from before that change for me? Are these chelper files compiled automatically, and is the resulting code picked up immediately? I did update Klipper before starting these jobs, and this might have been its first run.

dmbutyugin, 2020-04-30:
@KevinOConnor there is also (the same? stepcompress o=4 i=0 c=2 a=0: Invalid sequence) error mid-log, with full debug output available.

KevinOConnor, 2020-04-30:
I've seen a handful of reports similar to this over the last couple of months.  The symptom is a "stepcompress o=4 i=0 c=2 a=0: Invalid sequence" where the "o=" identifier matches the "extruder oid".  The reports seem to be from users that have pressure advance enabled and occur with a micro-controller that has been running for a week or more.

This is the first occurrence of the error I've seen after the "step+dir+step" filter was implemented, however.

My best guess at this point is that there is a rare numerical stability issue with the "smooth pressure advance" code.  (That code was added in late November).  

Unfortunately, I've been unable to reproduce the issue locally.  That makes it difficult to track down.  I'll try again over the next few days.

-Kevin

Maarten88, 2020-04-30:
My next 2 day long print succeeded, with nothing changed. I have restarted everything before restarting the failed print now.

I understand these are hard problems to find. Let me know if I can do something. Hope you can track this down!

Coffee0297, 2020-04-30:
we have had 2-3 guys on the discord with same issue. me included. made the issue but realiced i was on fandango's branch still. still after nov

kind regards Tonny

dmbutyugin, 2020-04-30:
@KevinOConnor I wanted to ask about this "step+dir+step" filter. The name suggests that it should resolve the issues with tmc2208 when 2 steps in the opposite direction are generated very close to each other. Does it also resolve some numeric stability issues? I have also seen that this filter supposedly resolved a similar issue on the scurve smoothing branch for one of the users, but I wonder if it happened because some issue was fixed or if the timings simply got different and some corner case in the step generation algorithm was not reached?

KevinOConnor, 2020-04-30:
> I wanted to ask about this "step+dir+step" filter. The name suggests that it should resolve the issues with tmc2208 when 2 steps in the opposite direction are generated very close to each other.

Yes, that is what the filter does.

> Does it also resolve some numeric stability issues?

No.  However, previous error reports seem to indicate that the "numerical instability" is generally manifest as rapid direction changes.  Specifically it seems as if a loss of precision in a kinematic calculation is causing the iterative solver to incorrectly detect backwards and forwards steps.  Thus, the step+dir+step filter was providing an inadvertent countermeasure.

> or if the timings simply got different and some corner case in the step generation algorithm was not reached?

The step+dir+step filter wont change the timings because it's applied after the iterative solver.  Unfortunately, I haven't been able to reproduce the issue so I don't yet know what timings lead to the root issue.

-Kevin

KevinOConnor, 2020-05-04:
I tracked down and fixed a numerical stability problem that I think is the cause of this issue (commit 7843c23b).  I recommend upgrading to the latest version of the software (see the FAQ for upgrade instructions).

-Kevin

closed 2020-05-11 by KevinOConnor