Klipper3d/klipper #2795: [DEVELOPER QUESTION] I2C: Set read timeout or Read without Write.

dianlight, 2020-04-28:
I'm implementing the python module to read the temperature from the HTU21D sensors via I2C. The problem is that from specifications the sensor responds asynchronously (with a delay of up to 30ms) to the read request.
As for how the features have been implemented, there is no i2c_read without writing and it is not possible to set the timeout to i2c_read.

Is there something that escapes me or did I just read the code?

If there is no functionality, I am available to develop a PR to add this type of functionality on the i2c bus.

L.

KevinOConnor, 2020-04-30:
To be honest, the Klipper i2c code isn't particularly flexible.  It was originally designed to setup stepper current via mcp4451 (and similar) dac chips.  It's since been used on a few other chips, but Klipper's error handling isn't great.  (Part of the problem, though, is that i2c hardware support is in general pretty fragile as well.)

That said, it should be possible to issue a Klipper i2c_read command with a zero length "reg" field.  That would be a read without any write.  I'm not familiar with the HTU21D so I'm not sure if that helps or not.

-Kevin

dianlight, 2020-04-30:
I checked the various implementations (I ignored architectures without i2c support or without i2c_read support.) and sending a write_len of 0 doesn't seem like a managed case to me:

- atsam ->  no control for len 0. data write channel open TWI_CR  set to START, no data sent, TWI_CT set to STOP
- avr -> no control for len 0. data write channel open, no data sent,  then data read channel open 
- linux -> io write with 0 len 

I don't have a hw avr or atsam to try but under linux the writing of a len 0 on at least a RPi 3B + can give problems (the i2c device become unresponsive).

Making a PR to handle the case at len ​​= 0 is quite simple and safe from regressions and would solve the problem only on linux where there is no timeout. On atsam and avr where the timeout 5000us the mcu would shutdown.From HTU21D datasheed read  has variable response time in the range of 4 / 20ms. The timeout window is too small.

If this is fine for you I would like a PR to better manage the case of len = 0 and add the support of i2c_read_if_data which in case of timeout does not bring the MCU shutdown but returns no data message .
So:
- Len = 0 case on all mcu with i2c_read.
- i2c_read_if_data would initially be supported on linux only. (I don't have hw available to test it on avr p atsam).

dianlight, 2020-04-30:
PR #2803 contains the driver with a len=0 patch for Linux_mcu.

if the i2c_read_if_data implementation is of interest, I can work on it.

KevinOConnor, 2020-05-11:
FWIW, I agree that all the architectures should treat a zero length write as a request for just a read.

-Kevin

KevinOConnor, 2020-05-24:
I'm going to close this as it looks like the original question was answered (the handling of read_length=0 requests in some architectures could be improved).

-Kevin

closed 2020-05-24 by KevinOConnor

NeilvdM, 2021-07-25:
> I'm implementing the python module to read the temperature from the HTU21D sensors via I2C. The problem is that from specifications the sensor responds asynchronously (with a delay of up to 30ms) to [...]

Hi @dianlight ,
I have been searching the web for help on my issue, which I hope is similar to what you were talking about here.
Im trying to connect a BME280 sensor to a RPi3B+
I have followed the examples in the Klipper documentation
I keep on ending with: MCU 'mcu' shutdown: i2c timeout.

For MCU, I have this in the Config file:
[mcu host]
serial: /tmp/klipper_host_mcu

[mcu]
#make sure this reflects your serial port you found when flashing klipper
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

and the BME280:
[temperature_sensor my_sensor]
sensor_type: BME280
gcode_id: AMB

another Example I found for the BME280, you have to include the Bus, but that gives me this error:

[temperature_sensor my_sensor]
sensor_type: BME280
gcode_id: AMB
i2c_address: 118
#   Default is 118 (0x76). Some BME280 sensors have an address of 119
#   (0x77).
i2c_mcu: mcu
i2c_bus: i2c.1
i2c_speed: 100000
#   See the "common I2C settings" section for a description of the
#   above parameters.

the Error:
Unknown i2c_bus 'i2c.1'