Klipper3d/klipper #2799: Connection to printer

patrickb08, 2020-04-29:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4551354/klippy.log)
[octoprint.log](https://github.com/KevinOConnor/klipper/files/4551355/octoprint.log)
[serial.log](https://github.com/KevinOConnor/klipper/files/4551361/serial.log)
Hello,

Having failed to install klipper with dwc2 on duet maestro, I decided to do it with octoprint but problem.
I cannot connect the printer to octoprint.
I have this in the terminal:

Connecting to: /dev/serial/by-id/usb-Klipper_sam4s8c_00313453464A4C4B3136303234303333-if00
Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
Unexpected error while connecting to serial port: /dev/serial/by-id/usb-Klipper_sam4s8c_00313453464A4C4B3136303234303333-if00 SerialException: '[Errno 11] Could not exclusively lock port /dev/serial/by-id/usb-Klipper_sam4s8c_00313453464A4C4B3136303234303333-if00: [Errno 11] Resource temporarily unavailable' @ comm.py:_openSerial:2691 (hook default)

BlackStump, 2020-04-29:
the klipper log looks fine, you problem is your config of octoprint.
The serial port that Octoprint connects to is /tmp/printer
Read the section "Configuring Octoprint to use Klipper"
https://www.klipper3d.org/Installation.html

bixicy, 2020-04-29:
Im also very new on Klipper.

Did you added /tmp/printer in your Octoprint Webfrontend?

patrickb08, 2020-04-29:
Thank you for answering me, I think I understand that / tmp / printer is a virtual printer and following the tutorial in the printer.cfg file you must put the port / serial / by-id / to be connected to the printer.

patrickb08, 2020-04-30:
No body for help !!!!!

BlackStump, 2020-04-30:
Do as I suggested and restart Klipper and then add the log from that session. The Original log was indicating you where trying to get Octoprint to connect to klipper's serial port to the mcu instead of the correct port that is /tmp/printer as per the link I send. I understand you are frustrated but you have to do your part for any help to come your way.

KevinOConnor, 2020-05-11:
I'm going to close this as it looks like the original question was answered (octoprint was not configured correctly).

-Kevin

closed 2020-05-11 by KevinOConnor