Klipper3d/klipper #2800: [Feature Request] Output-only bed mesh probing for use with bed visualizers

matthock, 2020-04-29:
Currently, mesh and tilt leveling settings are mutually exclusive. It would be useful to be able to enable them both at the same time, at least in the context of dumping information to the terminal - essentially, be able to do a "query-only" version of mesh level while tilt level is enabled for actual usage. The goal would be to use tilt for day to day usage, but you could still run the mesh for the Octoprint Bed Visualizer.

Another similar option would be to have a separate query-only mesh command for use by Bed Visualizer. This would also be useful even if you are using mesh leveling day to day, as there are some settings you would want for Bed Visualizer that are less useful for operation and vice versa (for example, a denser mesh without interpolation for the visualizer). This wouldn't even necessarily need a config section - the arguments (such as query area and density) could be passed as extended GCODE arguments to the command as long as a probe is configured.

GarthSnyder, 2020-04-29:
@matthock, could you expand a bit on the motivation behind this? If you're set up for mesh leveling, what are the factors that motivate you to avoid it for general use?

Coffee0297, 2020-04-29:
there allready is? im using ``BED_MESH_OUTPUT`` for octo, works great?
kind regards Tonny

matthock, 2020-04-30:
It's admittedly a bit of a niche use case - I do a lot of mods on the printer and it's useful to be able to run a dense mesh to verify that everything's still square. However, this takes a long time, so I prefer to run tilt for day to day printing for faster startup. Perhaps I'll look at it myself as it's likely niche enough to not be worth main developer time.

Arksine, 2020-05-01:
I think what you are requesting would be relatively simple to implement.  Something like this should work:
```diff
--- a/klippy/extras/bed_mesh.py
+++ b/klippy/extras/bed_mesh.py
@@ -86,7 +86,8 @@ class BedMesh:
         self.gcode.register_command(
             'BED_MESH_CLEAR', self.cmd_BED_MESH_CLEAR,
             desc=self.cmd_BED_MESH_CLEAR_help)
-        self.gcode.set_move_transform(self)
+        if not config.getboolean('diagnostic', False):
+            self.gcode.set_move_transform(self)
     def handle_ready(self):
         self.toolhead = self.printer.lookup_object('toolhead')
         self.bmc.handle_ready()
```

With that patch applied, add `diagnostic: True` to the [bed_mesh] section and it should allow you to probe and generate a mesh without transforming moves.

KevinOConnor, 2020-05-24:
Any further updates on this issue?  I'm guessing the original question was answered.

-Kevin

matthock, 2020-05-24:
You can close it if you wish. I don't think there's a ton of demand for it, I can always make a new one or ask for it to be reopened if I get around to implementing it.

KevinOConnor, 2020-06-01:
Okay, lets mark it closed until there is some further progress.

-Kevin

closed 2020-06-01 by KevinOConnor