Klipper3d/klipper #2805: Board not seen after flash

ibsmith04, 2020-04-30:
[klippyLog.zip](https://github.com/KevinOConnor/klipper/files/4560483/klippyLog.zip)

Hi,
I have been using klipper for about 3 months on an Ender 3 with 8bit melzi boardand love it. I decided to upgrade to an SKR 1.4 turbo with tmc2209's, more for the quiet drivers than anything else.

I started with a new install of octoprint, when I got to the klipper install I had trouble with
ls /dev/serial/by-id/*
as it said "no such file or directory". I did eventually get it to work as I hadn't powered up the SKR and got
/dev/serial/by-id/usb-marlinfw.org_Marlin_USB_Device_16FF800EAF3A81255C6BE1D9F50020C0-if00

I uploaded the firmware to the SKR by copying klipper.bin to firmware.bin on an SD card and booting the SKR from the card. That worked, in that the file was renamed FIRMWARE.COM

But I am now unable to connect to my new SKR board,
running ls /dev/serial/by-id/* just says no such file or directory, and "serial" just does not exist under /dev

The first time I installed klipper, in menuconfig, I saw an option to use USB rather than serial (I don't think that was there when I installed klipper in January), and I selected this option.

So wondering if this was the problem, I started all over again with a clean install of octoprint, then added klipper, and this time I did not select the USB instead of serial option.

I still couldn't get ls /dev/serial/by-id/* to work, but reasoned that after updating the firmware I should be able to connect. I updated the firmware in the same way, ie used an sd card.

But I still have the same problem, the pi cannot see the SKR board, /dev does not include a folder called serial.

I decided to try connecting to my old 8bit melzi board. I attached power and the usb cable only.
The pi immediately saw the board, and ls /dev/serial/by-id/* worked properly.

I then tried removing all wires from the SKR apart from power and usb. But still the pi could not see the SKR board.

I am hoping that it might be one of the parameters in menuconfig that has caused this, as I did manage to connect to the board prior to flashing the new firmware, what do you think ?

Thank you
Ian

ibsmith04, 2020-04-30:
I have just reflashed my SKR 1.4 back to firmware it came with (I had saved the FIRMWARE.CUR on the original sd card).
Now when I connect my pi to the SKR, and run ls /dev/serial/by-id/* , it works and shows me a device.
So at least I know it not a hardware issue, just have to work out what option to use in menuconfig.

ian

ibsmith04, 2020-05-01:
Got it working this morning. Tried several combinations in menuconfig, and ound one that works. Well so far it allows me to connect to my SKR, and klipper is not complaining about any of the setting in my cfg. The display is working, but is buzzing continuously. Off now to find out how to get rid of a buzzing screen

Ian

ibsmith04, 2020-05-01:
Have now got rid of the buzz, found I had to include
[output_pin beeper]
pin: P1.30
in my printer cfg

I have tested the heaters for hotend and bed and that all works nicely.
All I need to get working now are the bltouch and tmc2209's with sersorless homing.

ian

ibib, 2020-05-01:
@ibsmith04 I'm in the same situation as you are. Can you maybe share how you got it working? With wich settings and so on?

ibsmith04, 2020-05-01:
Hi ibib,

I have now got mine working correctly, in that everything moves in the correct direction, G28 work with sensorless endstops, the heaters work, I just have to calibrate everything now.

To get get a working firmware I had to use these settings in menuconfig
Low Level - NO
LPC176x
LPC1769
Smoothieware Bootloader YES
Use USB for connection YES

then copy klipper.bin to an sdcard
rename it to firmware.bin
switch off and disconnect SKR from pi
insert sdcard
turn on SKR
after 10 secs turn it off
put sdcard in pc, and check that file has been renamed to FIRMWARE.CUR

you should now be able to get the pi to see your SKR and  ls /dev/serial/by-id/* should give you the address required doe MCU section in printer.cfg

Good Luck

cheers
Ian

KevinOConnor, 2020-05-11:
Okay, I'm going to close this as it looks like a solution was found for the original issue.

-Kevin

closed 2020-05-11 by KevinOConnor