Klipper3d/klipper #2808: Flash without button press

lorenzing, 2020-05-01:
Hi,
I ask you if I can flash the board RUMBA32 without pressing the boot and reset keys, since the card is in an awkward position in the printer. 
I've seen the solution of Magokinbra in MK4duo ([https://github.com/Aus3D/RUMBA32/issues/16]) adding M9999 command. 
Thank you very much for helping.

Lorenzo

lorenzing, 2020-05-02:
Hi, 
the issue don't require lof file. 

Thank you again
Lorenzo

KevinOConnor, 2020-05-02:
Most boards can be flashed over usb, but it depends on the bootloader.  I don't know what bootloader is on that particular board.

-Kevin

Arksine, 2020-05-09:
The RUMBA32 appears to use a STM32F448, presumably they using the system bootloader.  Perhaps f8649b4ba9347bde03191bb0820ab23142a0b429 addressed this?

KevinOConnor, 2020-05-24:
I'm going to close this ticket as it looks like the conversation has concluded.

-Kevin

closed 2020-05-24 by KevinOConnor