Klipper3d/klipper #2809: Unexpected error while connecting to serial port: /tmp/printer SerialException: '[Errno 13] could not open port /tmp/printer: [Errno 13] Permission denied: '/tmp/printer'' @ comm.py:_openSerial:2691 (hook default)

joshschools, 2020-05-01:
I receive the following error in octoprint when I attempt to connect to the printer. The printer seems to be responding fine. Have restarted octoprint/printer/klipper with no success. This is a new install of Klipper on an Ender 3/Raspberry PI 3

● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated)
   Active: active (running) since Fri 2020-05-01 08:02:53 CDT; 21min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 1434 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCESS)
    Tasks: 4 (limit: 2077)
   Memory: 10.2M
   CGroup: /system.slice/klipper.service
           └─1439 /home/jschools/klippy-env/bin/python /home/jschools/klipper/klippy/klippy.py /home/jschools/printer.cfg -l /tmp/klippy.log

may 01 08:02:53 octopi systemd[1]: Starting LSB: Klipper daemon...
may 01 08:02:53 octopi klipper[1434]: Starting klipper: klipper.
may 01 08:02:53 octopi systemd[1]: Started LSB: Klipper daemon.

total 64
drwxrwxrwt  8 root     root      4096 may  1 08:11 .
drwxr-xr-x 21 root     root      4096 sep 25  2019 ..
drwxrwxrwt  2 root     root      4096 may  1 07:56 .font-unix
drwxrwxrwt  2 root     root      4096 may  1 07:56 .ICE-unix
-rw-r--r--  1 jschools jschools 32236 may  1 08:02 klippy.log
lrwxrwxrwx  1 jschools jschools    10 may  1 08:02 printer -> /dev/pts/0
lrwxrwxrwx  1 jschools jschools    10 may  1 07:56 printer.old -> /dev/pts/0
drwx------  3 root     root      4096 may  1 07:56 systemd-private-a279ff6020ce4992bdcc39172d6e97f4-systemd-timesyncd.service-OlYBx2
drwxrwxrwt  2 root     root      4096 may  1 07:56 .Test-unix
drwxrwxrwt  2 root     root      4096 may  1 07:56 .X11-unix
drwxrwxrwt  2 root     root      4096 may  1 07:56 .XIM-unix

[klippy.log](https://github.com/KevinOConnor/klipper/files/4564300/klippy.log)

joshschools, 2020-05-01:
Weird. When I pointed octoprint to /dev/pts/0 it connected. Will this work?

Connecting to: /dev/pts/0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x7006ea50, open=True>(port='/dev/pts/0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Send: N0 M110 N0*125
Changing monitoring state from "Connecting" to "Operational"
Recv: ok
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.8.0-449-gb6ebf5cb FIRMWARE_NAME:Klipper

KevinOConnor, 2020-05-02:
This appears to be the same root cause as issue #1902.  However, I have not seen this report on a standard octopi installation.  What image and machine did you use for the installation?

-Kevin

joshschools, 2020-05-03:
> This appears to be the same root cause as issue #1902. However, I have not seen this report on a standard octopi installation. What image and machine did you use for the installation?
> 
> -Kevin

I installed Klipper via CLI on the rpi3. Octopi was previously installed months ago. I have been using Marlin up until now. The printer is an Ender 3 Pro. I reviewed #1902 that appears to be the issue here. I am also on Buster on my rpi3.

KevinOConnor, 2020-05-24:
Okay, I'm going to close this issue as a dup of #1902 .

-Kevin

closed 2020-05-24 by KevinOConnor