Klipper3d/klipper #2810: Yet another unable to calibrate delta

theccm, 2020-05-01:
Hello,

I have a anycubic Kossel, pulley version, and facing to a problem I am unable to solve.
when moving at the center , going to Z=0 give a good position to the nozzle, I mean it is just above the head. But when moving outside near the tower I have +0.5mm -0.4mm depending X/Y pos. So I am unable to print for example the enhanced delta calibration object.
But not problem with small objects (40x40 for example)

So far what I can say and did :
- the build quality is good, The towers have square angles (btw what is the tolerance, 1° , 5° ?)
- basic tests done (endstop test, motors tests, etc)
- I'm using TMC2209 so I'm using phase probe for endstop
- arms length does have all the same value : 217mm
- delta radius is ok in config file : 95mm
- I'm using paper test method
- I ran DELTA_CALIBRATE method=MANUAL, did TESTZ 6/7 times, then SAVE_CONFIG
- and then I saw that problem
- I did also BED_MESH_CALIBRATE command but still the same issue.

during PRINT I tried SET GCODE_OFFSET Z_ADJUST=x MOVE=1 but of course when I'm ok at the outside I'm too far at the center.

Also when moving near the outside, the head and effector are not vertical is it normal ?
[image]

[image]

thanks for your help. As english is not my natural language maybe I misunderstood something.

Last thing I didn't understand difference between DELTA_CALIBRATE and BED_MESH_CALIBRATE macro.

Hywelmartin, 2020-05-01:
you need to fix the arms... are you missing spacers or is there worn out links?

BlackStump, 2020-05-01:
your delta radius is wrong, your effective is moving on a concave manner

mak5555, 2020-05-03:
> Also when moving near the outside, the head and effector are not vertical is it normal ?

NOT AT ALL !
The effector should be always parallel to the bed
You have a MECHANICAL issue

With a pulley version, it should be bad sliders with the top end of the arms moving away from the towers.
Try to tighten the sliders screw.
You can install better sliders with a backplate, for example 
https://www.thingiverse.com/thing:2512391

BED_MESH_CALIBRATE is usefull with a not so flat Bed .
I don't use it with a glass bed.

theccm, 2020-05-03:
Hello,

I made signifiant progress.

> you need to fix the arms... are you missing spacers or is there worn out links?

you are right. I have the first Kossel version and I didn't know it had design issues. I printed 5mm spacers, for the effector and for the carriages.

> NOT AT ALL !
The effector should be always parallel to the bed
You have a MECHANICAL issue

Yes I readed many docs since 2 days and the above fix did the trick. Also delta radius has ben set to 97.4

and the second issue, I'm feeling stupid but the C endstop was 2cm under the others ! Apart from that I think that one very important calibration - and I did not see it in klipper docs or elsewhere - is to go to towers coordinates and check that Z=0 is just above the bed. In my case I had few differences even if the endstop positions where good and phase detection enabled.

I have now 0.1 mm difference with one side but I can now run enhanced delta calibration object so I will check if it will solde this last difference.

thanks for your answers anyway.

KevinOConnor, 2020-05-11:
I'm going to close this as it looks like a solution was found for the original issue.

-Kevin

closed 2020-05-11 by KevinOConnor