Klipper3d/klipper #2811: [bed_mesh] crash the Printer

darksoon, 2020-05-01:
Hello everyone,

here's the problem. When i set the bed mesh in my config the printer is not reachable anymore. If i leave out this setting the printer starts as usual and i can also run a G28. This "bug" only occurs no matter at which Position I insert the [bed_mesh]. I have included the current log. I hope someone can help me.

greetings

[klippy.log](https://github.com/KevinOConnor/klipper/files/4564830/klippy.log)

Arksine, 2020-05-01:
It appears that Repetier is sending a gcode to Klipper while your MCU is trying to connect and bed_mesh isn't handling well.  I just created a PR for a fix.

darksoon, 2020-05-01:
Hey,
This fix works perfekt thank you 😊

closed 2020-05-02 by darksoon