Klipper3d/klipper #2817: Internal error on command:"G1"

TheBigF1sh, 2020-05-02:
[New Text Document.txt](https://github.com/KevinOConnor/klipper/files/4567961/New.Text.Document.txt)

Hi 

I think I can see a few issues around this but states code upgraded to resolve - however this is the second time I have now had a cancelled print due to the G1 error.

Any help you can provide to enable me to circumvent would be greatly appreciated.

KevinOConnor, 2020-05-02:
The root cause of your issue is a "Heater extruder not heating at expected rate" issue.  Your extruder temperature is fluctuating wildly - that's usually an indicator of a faulty thermistor or its wiring.

-Kevin

TheBigF1sh, 2020-05-02:
Thanks for the speedy response. Ill take a look at the wiring.

closed 2020-05-02 by TheBigF1sh