Klipper3d/klipper #2819: strange behavior pid calibration

ricoelmostico, 2020-05-02:
When I try a pid calibration on my extruder I got a weird behavior and bad result.
I used octoklipper plugin.
It sends this command:

"Send: PID_CALIBRATE HEATER=extruder TARGET=245"

after calibration it return this message:

"Recv: // PID parameters: pid_Kp=22.019 pid_Ki=36.698 pid_Kd=3.303
Recv: // The SAVE_CONFIG command will update the printer config file
Recv: // with these parameters and restart the printer."

Seems like the calibration script has done only one spike:
[image]

log file after an m112:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4569339/klippy.log)

ricoelmostico, 2020-05-11:
@KevinOConnor sorry I'm kind of desperate..
I did many fresh install of octoprint and klipper (from the book, looking at octoprinter install procedure and install.md from klipper), re-written my cfg file, mesured my heater cartradge, changed my thermistor, checked my wires..
Can you have a look at my log file? It's gibberish for me x)

KevinOConnor, 2020-05-11:
Your thermistor readings are changing wildly over a very short period of time.  This is often caused by a malfunctioning thermistor or a failure in the wiring.  As a result of the dramatic temperature swings, the PID_CALIBRATE command is not able to obtain meaningful results.  I recommend you triple check the thermistor wiring.

-Kevin

ricoelmostico, 2020-05-12:
yes indeed it is that, thank you !

closed 2020-05-24 by KevinOConnor