Klipper3d/klipper #2821: Replicape: Servo moves uncontrolled during startup or when servo is disabled

Szeker, 2020-05-03:
On the Replicape board, 

1.) There is a short period of time during startup, when the servo is moving uncontrolled (going crazy...).

2.) In case the servo is disabled by the SET_SERVO command, the servo keeps behaving crazy continuously.

"Servo goes crazy" means, it moves uncontrolled far out of it's normal operating angle range.

At the moment there is a general issue with the servo on Replicape (#2746), so I experienced the listed behavior during testing the branch of @jannau - fixing the general servo on Replicape issue:

https://github.com/KevinOConnor/klipper/pull/2746

Assumptions:
1.) During startup: the PWM output buffer is enabled (via SPI) before the PWM pin of the BBB is configured correctly

2.) During servo disabled: the PWM pin of the BBB is configured to a non servo compatible mode, while the PWM output buffer remains enabled.

Klippy.log attached.

[klippy_log.zip](https://github.com/KevinOConnor/klipper/files/4570242/klippy_log.zip)

KevinOConnor, 2020-05-11:
@jannau - any comments on this?

-Kevin

KevinOConnor, 2020-07-10:
Alas, I don't have any way to help with these Replicape + Linux issues.  Did this problem occur on the recommended Linux image?  Perhaps @profdryoman or @goeland86 could help?

-Kevin

goeland86, 2020-07-10:
@Szeker can you remind me - is your Replicape B3 or B3A? The B3 had some floating pins that may account for that in addition to the buzzing steppers, but if it's also present on B3A then maybe we need to look at making some custom uboot additions that sets the pins to the right state before Linux takes over.

Szeker, 2020-07-11:
@goeland86 I have a B3A board.

goeland86, 2020-07-12:
Ok, thanks. I'll see what we can do to fix that.

Szeker, 2020-07-14:
Regarding the servo, I made a quick measurement on the servo pin with an oscilloscope, and I saw the following:

1.) During startup of the system, first there's 0V on the servo pin for a while, which makes the servo just keeping it's last position, but than there is a short period of time, when a periodic 320ns pulse appears on the servo pin.  That 320ns pulse is what makes the servo crazy. After that short period with those 320ns pulses, the correct signal is set on the servo output pin, and the servo works as expected - correctly.

2.) The same applies when I disable the servo: a continuous, periodic 320ns pulse appears on the servo pin. The period is 20ms - which is the same as during normal/correct operation of the servo (= when the servo is enabled).I don't know the exact timers of the BBB, but my guess would be, that the timer module can not output exact 0% (or 100%) duty cycle / pulse width. This happens on many uC's timer module.
So probably it outputs a 1 clock cycle high pulse in each 20ms period - instead of a steady 0 state.
In these cases, the port pin typically have to be set manually to output port mode (instead of timer/periferel mode) + the target level have to be also set to the port pin "manually".But that's just guessing as I don't know the details of the timer module on the BBB...

KevinOConnor, 2020-07-16:
If you make the following change to the linux mcu code, does it improve the results?

```diff
--- a/src/linux/hard_pwm.c
+++ b/src/linux/hard_pwm.c
@@ -100,15 +100,10 @@ fail:
     shutdown("Unable to config pwm device");
 }
 
-
 void gpio_pwm_write(struct gpio_pwm g, uint16_t val)
 {
     char scratch[16];
     uint32_t duty_cycle = g.period * (uint64_t)val / MAX_PWM;
-    snprintf(scratch, sizeof(scratch), "%u", duty_cycle);
-    if (g.fd != -1) {
-        write(g.fd, scratch, strlen(scratch));
-    } else {
-        report_errno("pwm set duty_cycle", g.fd);
-    }
+    int len = snprintf(scratch, sizeof(scratch), "%u", duty_cycle);
+    write(g.fd, scratch, len + 1);
 }
```

-Kevin

Szeker, 2020-07-17:
I tried the patch, but unfortunately there's still that 320ns periodic pulse generated on the servo PWM pin in case the servo is disabled (and during startup for a short period of time).

I even tried the code with fixed 0 duty cycle - just for test purposes, but the 320ns pulse is always there.
Therefore I still consider it as a HW limitation of the PWM module of the BBB, but I have to check it more in details.

KevinOConnor, 2020-07-19:
That's certainly odd.  If you stop Klipper and manually setup the pwm with something like:
```
echo 20000000 > /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/period
echo 0 > /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/duty_cycle
echo 1 > /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/enable
```
Do you still see the issue?

If so, that sounds like a quirk in the beaglebone pwm implementation.  Or perhaps, the code is expected to write "0" to the .../pwm/enable file on a zero pulse length.  @jannau - do you know if that is true?

-Kevin

Szeker, 2020-07-19:
@KevinOConnor : what user / how can I send directly the mentioned PWM configs? I've got "permission denied" by sending them manually.

Szeker, 2020-07-19:
@KevinOConnor 

Regarding the pwm, I found the following:

- The resolution of the PWM seems 320ns, which means the duty cycle can be adjusted by 320ns steps.
- Mapping of duty_cycle: 
     - 0-319     => 320ns pulse on output
     - 320-639 => 640ns pulse on output
     - 640-959 => 960ns pulse on output
     - etc...

- The minimum pulse with what the PWM module can generate is 320ns (in case the module is enabled) = the resolution of the PWM, as stated earlier

- If the PWM module is disabled (enable = 0), output stays stable low

**Based on the above measurements, my proposal is to disable the PWM in case we want to disable it - instead of setting the duty cycle to 0.**

KevinOConnor, 2020-07-20:
I don't have the hardware, but that change doesn't seem hard.  If you're comfortable making changes to the software you could try the following:
```diff
diff --git a/src/linux/gpio.h b/src/linux/gpio.h
index e6ef86bd..e2a39be0 100644
--- a/src/linux/gpio.h
+++ b/src/linux/gpio.h
@@ -37,7 +37,7 @@ void spi_transfer(struct spi_config config, uint8_t receive_data
                   , uint8_t len, uint8_t *data);
 
 struct gpio_pwm {
-    int fd;
+    int duty_fd, enable_fd;
     uint32_t period;
 };
 struct gpio_pwm gpio_pwm_setup(uint32_t pin, uint32_t cycle_time, uint16_t val);
diff --git a/src/linux/hard_pwm.c b/src/linux/hard_pwm.c
index cfe8b436..3e5bb6ee 100644
--- a/src/linux/hard_pwm.c
+++ b/src/linux/hard_pwm.c
@@ -77,20 +77,18 @@ struct gpio_pwm gpio_pwm_setup(uint32_t pin, uint32_t cycle_time, uint16_t val)
         report_errno("pwm duty_cycle", fd);
         goto fail;
     }
-
-    g.fd = fd;
-    gpio_pwm_write(g, val);
+    g.duty_fd = fd;
 
     // enable PWM
     snprintf(filename, sizeof(filename), pwm_path, chip_id, pwm_id, "enable");
     fd = open(filename, O_WRONLY|O_CLOEXEC);
     if (fd == -1) {
-        close(g.fd);
+        close(g.duty_fd);
         report_errno("pwm enable", fd);
         goto fail;
     }
-    write(fd, "1", 2);
-    close(fd);
+    g.enable_fd = fd;
+    gpio_pwm_write(g, val);
 
     return g;
 
@@ -105,10 +103,7 @@ void gpio_pwm_write(struct gpio_pwm g, uint16_t val)
 {
     char scratch[16];
     uint32_t duty_cycle = g.period * (uint64_t)val / MAX_PWM;
-    snprintf(scratch, sizeof(scratch), "%u", duty_cycle);
-    if (g.fd != -1) {
-        write(g.fd, scratch, strlen(scratch));
-    } else {
-        report_errno("pwm set duty_cycle", g.fd);
-    }
+    int len = snprintf(scratch, sizeof(scratch), "%u", duty_cycle);
+    write(g.duty_fd, scratch, len + 1);
+    write(g.enable_fd, val ? "1" : "0", 2);
 }
```

-Kevin

Szeker, 2020-07-20:
The solution works perfectly!
No glitches on the oscilloscope, the servo doesn't go crazy when I disable it, no any movement during startup!

Perfect!

Thanks very much for the responsiveness!
You're a real Master!

KevinOConnor, 2020-07-21:
Great.  I committed this change (commit 5f2cb543).

-Kevin

Szeker, 2020-07-21:
The commit 5f2cb5436cf545c2d5efba7994f458b98644ef6f solved the issue and works perfectly.

Thanks!

closed 2020-07-21 by Szeker