Klipper3d/klipper #2822: TriGorrila board.

Hairyloon, 2020-05-03:
Did I not post my post about my board blowing up or did it get taken down? No matter, I've got a new board now, but I can't find it listed in the config folder.
Is a TriGorrilla 1.1, which I believe comes from AnyCubic. There are two AnyCubic files: i3 and 4max: which is it more likely to be?
Or do I have to go through all the pins manually?

Hywelmartin, 2020-05-03:
believe it's more or less a generic-ramps.cfg

ruevs, 2020-05-03:
The Anycubic deltas use the TriGorilla board.
For example:
https://github.com/KevinOConnor/klipper/blob/master/config/printer-anycubic-kossel-plus-2017.cfg

Hairyloon, 2020-05-03:
> believe it's more or less a generic-ramps.cfg

Yes, but the question is one of pin assignments or aliases.

Hairyloon, 2020-05-03:
> The Anycubic deltas use the TriGorilla board.
> For example:
> https://github.com/KevinOConnor/klipper/blob/master/config/printer-anycubic-kossel-plus-2017.cfg

Ah, good stuff. Thank you.
Is it Ok to leave this open for now, in case I have trouble translating it into Cartesian?

Hywelmartin, 2020-05-03:
[TriGoRiLLa_PINs.pdf](https://github.com/KevinOConnor/klipper/files/4571448/TriGoRiLLa_PINs.pdf)
here is the pinout

schnello, 2020-05-03:
>Is it Ok to leave this open for now, in case I have trouble translating it into Cartesian?

Use the i3config. Same Board but Cartesian.
https://github.com/KevinOConnor/klipper/blob/master/config/printer-anycubic-i3-mega-2017.cfg

Hairyloon, 2020-05-03:
Cool, thank you.
Do we know that that's the same board?  I'd looked, but found nothing to confirm it.

schnello, 2020-05-04:
Again. The i3 Mega uses the TriGorrilla 1.x. This is a fact (I had this printer for two years)

Hairyloon, 2020-05-04:
Thank you.
Do you think it is too much to ask for them to put a note to that effect in the files?
They seemed pretty common in the shops & it would make it a lot easier for the next chap.

schnello, 2020-05-04:
Normaly you purchase this kind of board only for the specific printer.

Hairyloon, 2020-05-04:
> Normaly you purchase this kind of board only for the specific printer.

Normally you wouldn't have a printer with a blown board.

Hairyloon, 2020-05-04:
> > Is it Ok to leave this open for now, in case I have trouble translating it into Cartesian?
> 
> Use the i3config. Same Board but Cartesian.
> https://github.com/KevinOConnor/klipper/blob/master/config/printer-anycubic-i3-mega-2017.cfg

The i3 has no mention of a display in the config. Does it not have one, or is it worked differently?
Either way, can I get the display from the Anet A8 to work through the configuration, or do I need to rewire it?

Hywelmartin, 2020-05-04:
```
# Common EXP1 / EXP2 (display) pins
[board_pins]
aliases:
    # Common EXP1 header found on many "all-in-one" ramps clones
    EXP1_1=ar37, EXP1_3=ar17, EXP1_5=ar23, EXP1_7=ar27, EXP1_9=<GND>,
    EXP1_2=ar35, EXP1_4=ar16, EXP1_6=ar25, EXP1_8=ar29, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=ar50, EXP2_3=ar31, EXP2_5=ar33, EXP2_7=ar49, EXP2_9=<GND>,
    EXP2_2=ar52, EXP2_4=ar53, EXP2_6=ar51, EXP2_8=ar41, EXP2_10=<RST>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "spi"
    # Note, some boards wire: EXP2_8=<RST>, EXP2_10=ar41

# See the sample-lcd.cfg file for definitions of common LCD displays.
```

https://github.com/KevinOConnor/klipper/blob/master/config/sample-lcd.cfg

Hairyloon, 2020-05-04:
Tidy, thank you.

Hywelmartin, 2020-05-04:
as the a8 uses analog for buttons you will have to change  some pins

https://github.com/KevinOConnor/klipper/blob/9a92b346fd81386c54464b45d8310dab972b6bbd/config/printer-anet-a8-2017.cfg#L81

Hairyloon, 2020-05-07:
Well, bits of it are working.
Some of it I just need to work through, but the main item of concern now is the Z axis doesn't seem to want to home properly: it doesn't start by going up, it winds down to the endstop, then starts whining, then after a moment the motors begin clicking backwards and forwards.

As a minor point, the Y axis wants to home in the opposite direction. I expect I can sort that out if I RTFB, but I have to go shopping now, so I thought I'd ask while I'm here...
ETA: pling the direction pin: that does it. :)

[klippy.log](https://github.com/KevinOConnor/klipper/files/4594908/klippy.log)

Hairyloon, 2020-05-07:
OK, changing the step size has improved things: it now homes, but it still does the whining and the back and forth...

Hairyloon, 2020-05-08:
> 
> 
> [TriGoRiLLa_PINs.pdf](https://github.com/KevinOConnor/klipper/files/4571448/TriGoRiLLa_PINs.pdf)
> here is the pinout
> 
> E1 is
> #[extruder1]
> #step_pin: ar36
> #dir_pin: ar34
> #enable_pin: !ar30
> 
> I think...

Just a small point but that picture is wrong: it has E1 labelled as "Z motor" with a blank space where the 2nd Z motor is.

Trivial I know, but it just gave me a moment of "Have I wired it up wrong?? Is the problem so simple" and a dash out to check...

Hairyloon, 2020-05-08:
OK the printer is working now, but I am yet to get the display to work.
To be blunt, I don't understand the advice that's been given.  
I thought I did, tried it and it didn't work.
I've not tried very hard yet: getting it printing was a bigger priority and I don't want to tinker while it's printing.

KevinOConnor, 2020-05-24:
I'm going to close this ticket as it looks like a solution was found for the original issue.

-Kevin

closed 2020-05-24 by KevinOConnor

Hairyloon, 2020-05-24:
> I'm going to close this ticket as it looks like a solution was found for the original issue.
> 
> -Kevin

The original issue yes, but it seems there is ambiguous documentation in relation to the board and it would be quite good to clear that up: specifically that this diagram https://github.com/KevinOConnor/klipper/files/4571448/TriGoRiLLa_PINs.pdf
shows the Z1 motor connecting to the E1 driver and it's not clear if that is necessary, recommended or in error.
I'm guessing there is documentation to go with the picture, but I've not been able to find it.

Also I wasn't able to get the display working: I got confused by the instructions and the wiring diagram.
It wasn't a priority.

schnello, 2020-05-24:
>Shows the Z1 motor connecting to the E1 driver and it's not clear if that is necessary, recommended or in error.

If you want use a second separate controlled Z Motor... the only option is E1.

Hairyloon, 2020-05-24:
> > Shows the Z1 motor connecting to the E1 driver and it's not clear if that is necessary, recommended or in error.
> 
> If you want use a second separate controlled Z Motor... the only free port is E1.

Obviously so, but why would I want to?
I did, because I was having trouble caused by failing to tune the driver boards, and I've left it like that because I have no need to change it, and I think I'd need to retune the driver, but at some point I may want a second extruder, so it would be helpful to know before then.

schnello, 2020-05-24:
>Obviously so, but why would I want to?

To automaticaly aign the z axis (left and right) with to motors and two endstops. Works great with the i3 MEGA.  
[image]

Hairyloon, 2020-05-26:
> 
> 
> > Obviously so, but why would I want to?
> 
> To automaticaly aign the z axis (left and right) with to motors and two endstops. Works great with the i3 MEGA.

Well yes, obviously, though I think levelling with respect to the bed is more useful than to the endstops... which is something that I think I spotted somewhere in the instructions, but considering how often I expect to need to do that and how easily done it is by hand, I'm not convinced of the merit, where I can certainly see some potential use for a second extruder (or something else).

Hairyloon, 2020-08-04:
> Well yes, obviously, though I think levelling with respect to the bed is more useful than to the endstops... which is something that I think I spotted somewhere in the instructions, but considering [...]

Yeah, OK. Having tried it I totally see your point...

Would anybody mind if reopened this thread for a bit?
I couldn't get the display to work at all and I'm looking for guidance.
I don't know if I've got it wired up wrong, got the pin settings wrong or even if I'd managed to kill the display board altogether.
Is the stock 2004 LCD off the Anet A8 and a Trigorilla board.
Any pointers?

Hywelmartin, 2020-08-04:
https://github.com/KevinOConnor/klipper/blob/fe84968bdcb1d17455ca6c2ffb9ed8d209dc85ed/config/printer-anycubic-kossel-plus-2017.cfg#L101

Hairyloon, 2020-08-04:
> 
> 
> https://github.com/KevinOConnor/klipper/blob/fe84968bdcb1d17455ca6c2ffb9ed8d209dc85ed/config/printer-anycubic-kossel-plus-2017.cfg#L101

Thank you, and yes, that's what I've got, though mine goes through the [board_pins] aliases.  
I've tried changing it, in case the aliases are somehow not working and it's still dead.
Looks like it may be the LCD...