Klipper3d/klipper #2823: Question/Feature Delta height by homing

PuddingBaer91, 2020-05-03:
Hi,

Klipper works well for me but i have a question/idea:

Why we can't set the z height or z=0 position for a delta printer by just homing the head. If the bed and nozzle is heated and the nozzle is sitting on the bed i would just do a home. It could just count the distance til it is homed. This should be the z height... Or i am missing sonething here?!?

Thank you

whoim2, 2020-05-03:
Analog G33 P1 in marlin? Autoheight from probe? Good idea.

KevinOConnor, 2020-05-03:
When a stepper motor is powered on, it "snaps" to a position based on the "phase" it is set to.  There's a little bit of background on this at https://www.klipper3d.org/Endstop_Phase.html .  There is no way for the software to know how much distance the head moves when the steppers "snap" into position and this distance is commonly around +-0.4mm .  To get good bed adhesion and general print stability we really want to measure the Z to a precision of around 25um (.025mm).

So, the mechanism you suggest might be a useful starting point for users that don't have any idea what their arm lengths, radius, or endstop positions are.  However, the full DELTA_CALIBRATE procedure is needed to get sufficient accuracy for actual printing.

-Kevin

PuddingBaer91, 2020-05-04:
Thank you for the information. Didn't know that snap-thing...

closed 2020-05-04 by PuddingBaer91