Klipper3d/klipper #2825: New extras module for loading and saving arbitrary variables to disk

dushyantahuja, 2020-05-04:
I've created a new extras module for loading and saving arbitrary variables to a separate file on the disk, so that the variable values can be retained across restarts. The load and save work on my testing fork:

https://github.com/dushyantahuja/klipper/blob/master/klippy/extras/save_variables.py

It needs a config section with the filename of the new file
[save_variables]
filename: variable.cfg

Variables can be saved using SAVE_VARIABLE VARIABLE=temp2 VALUE=23
and loaded using LOAD_VARIABLE VARIABLE=temp2

However, I've just started coding in Python, and I'm not sure what needs to be done to actually be able to use the variables once they're loaded. As an example, what I would like to do is:

1. Set a variable called 'parked' that tells my macro that the filament is parked -- Can be done using SAVE_VARIABLE VARIABLE=parked VALUE=1
2. Now, I want to use this variable in another macro to check if the filament is parked, and if not park it. The plan is to use LOAD_VARIABLE VARIABLE=parked -- the current code is able to read the variable value from the file, but I don't know how this can be used in the macro. Any help would be appreciated.

dushyantahuja, 2020-05-04:
#2761

KevinOConnor, 2020-05-05:
Typically, one would create a `get_status()` method that returns a Python dictionary.  Once that is done, it should be possible to use the contents of that dictionary in a macro - for example: `{printer.save_variables.my_variable}`.  Take a look at klippy/extras/servo.py as an example.

Not sure you need a LOAD_VARIABLE command - it sounds like you could just export all saved variables in a get_status() method.

-Kevin

dushyantahuja, 2020-05-05:
Thanks Kevin, I think I'm nearly there now. Have updated the code with a get_status() method. It is returning a dict directly. 

https://github.com/dushyantahuja/klipper/blob/3fc03a87852f0bcd4293cfb56d3a723a16a03624/klippy/extras/save_variables.py#L53

Unfortunately, when I try to load one of the variables, I get the following error:

Recv: !! Error evaluating 'gcode_macro TESTVARIABLES:gcode': UndefinedError: 'dict object' has no attribute 'temp1'

TESTVARIABLES is defined as:
[gcode_macro TESTVARIABLES]
gcode:
  {printer.gcode.action_respond_info(printer.save_variables.temp1)}

Relevant traceback from the log is:

Error evaluating 'gcode_macro TESTVARIABLES:gcode': UndefinedError: 'dict object' has no attribute 'temp1'
Traceback (most recent call last):
```
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 61, in render
    return str(self.template.render(context))
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "<template>", line 2, in top-level template code
  File "/home/pi/klipper/klippy/gcode.py", line 124, in _action_respond_info
    self.respond_info(msg)
  File "/home/pi/klipper/klippy/gcode.py", line 332, in respond_info
    lines = [l.strip() for l in msg.strip().split('\n')]
UndefinedError: 'dict object' has no attribute 'temp1'
Error evaluating 'gcode_macro TESTVARIABLES:gcode': UndefinedError: 'dict object' has no attribute 'temp1'
```
I've attached the log file and the variables.cfg file 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4583858/klippy.log)
[variable.cfg.zip](https://github.com/KevinOConnor/klipper/files/4583861/variable.cfg.zip)

As far as I understand, gcode.py is trying to split the response using \n - however the returned dict does not have a \n; instead it has commas to split various items:

{'temp2': '23', 'temp1': '4044', 'temp': '123'}

Apologies for bothering you again, but I'm a newbie in python, and am stuck here.

KevinOConnor, 2020-05-05:
Not sure, but I'd recommend adding logging statements to your code to track it down.  For example, add `logging.info("allVariables='%s'", repr(self.allVariables))` to the top of get_status().

Also, using the "action callbacks" is confusing things for you - try this in the gcode macro instead: `ECHO VAL={printer.save_variables.temp1}` .  You may also wish to just test with `ECHO VAL="{printer.save_variables}"`.

-Kevin

dushyantahuja, 2020-05-06:
Thanks for this - it's working now. I'll add some documentation and submit a pull request.

closed 2020-05-06 by dushyantahuja

dushyantahuja, 2020-05-06:
#2839

Goodfeat, 2020-05-14:
I tried adding the save_variables.py script to the extra folder. tried to create a configuration

[save_variables]
filename: variable.cfg

after that, the host cannot connect to the printer

07:11:05 Internal error during connect: global name 'error' is not defined

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready
07:11:05 Internal error during connect: global name 'error' is not defined

LOG
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
    self._read_config ()
  File "/home/pi/klipper/klippy/klippy.py", line 129, in _read_config
    self.try_load_module (config, section_config.get_name ())
  File "/home/pi/klipper/klippy/klippy.py", line 118, in try_load_module
    self.objects [section] = init_func (config.getsection (section))
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 52, in load_config
    return SaveVariables (config)
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 17, in __init__
    self.loadVariables ()
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 29, in loadVariables
    except error as e:
NameError: global name 'error' is not defined
Internal error during connect: global name 'error' is not defined

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready

dushyantahuja, 2020-05-14:
I've just updated the file - while removing the global error variable, I had missed removing it from one of the blocks. 

Please try with: https://github.com/dushyantahuja/klipper/blob/master/klippy/extras/save_variables.py

Also, you need to add 
[save_variables]
filename: ~/variable.cfg

Goodfeat, 2020-05-14:
I'm probably doing something wrong, sorry

> If a [save_variables] config section has been enabled,
> `SAVE_VARIABLE VARIABLE=<name> VALUE=<value>` can be used to save the variable
> to disk so that it can be used across restarts. All stored variables are loaded
> into the printer.save_variables dict at startup and can be used in gcode macros.
> 
> As an example, it could be used to save the state of 2-in-1-out hotend
> and when starting a print ensure that the active extruder is used, instead of T0
> 

```
[save_variables]
filename: ~ / variable.cfg
SAVE_VARIABLE VARIABLE = extruder VALUE = 0
#SAVE_VARIABLE VARIABLE = <extr> VALUE = <0>
# VARIABLE = extr VALUE = 0
```

07:33:28 Internal error during connect: No section: 'Variables'
Unable to parse existing variable file

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready
07:33:28 Internal error during connect: No section: bVariables

log
No section: 'Variables'
Unable to parse existing variable file
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 25, in loadVariables
    self.allVariables = dict (self.variablefile.items ('Variables'))
  File "/usr/lib/python2.7/ConfigParser.py", line 642, in items
    raise NoSectionError (section)
NoSectionError: No section: 'Variables'
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
    self._read_config ()
  File "/home/pi/klipper/klippy/klippy.py", line 129, in _read_config
    self.try_load_module (config, section_config.get_name ())
  File "/home/pi/klipper/klippy/klippy.py", line 118, in try_load_module
    self.objects [section] = init_func (config.getsection (section))
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 57, in load_config
    return SaveVariables (config)
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 14, in __init__
    self.loadVariables ()
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 29, in loadVariables
    raise self.gcode.error (msg)
CommandError: No section: 'Variables'
Unable to parse existing variable file

dushyantahuja, 2020-05-14:
You shouldn't have spaces in the filename - it should be ~/variable.cfg

dushyantahuja, 2020-05-14:
Also, have you got the latest file from the link I provided above?

Goodfeat, 2020-05-14:
I have no spaces in the original file, so it was formatted when copying
yes, I copied the code and replaced it in the file that you suggested
[config and save variable.zip](https://github.com/KevinOConnor/klipper/files/4626521/config.and.save.variable.zip)

dushyantahuja, 2020-05-14:
Thanks for testing - it's not your fault, it's me not knowing enough python and diving deep :-)
The error comes up because the file variable.cfg does not exist. Let me update the code, and I'll get back to you

Goodfeat, 2020-05-14:
Thank you too for your work,

I updated the code, it changed something in the logs

> 08:10:56 Internal error during connect:
> Unable to parse existing variable file
> 
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Printer is halted
> Klipper state: Not ready
> 08:10:56 Internal error during connect:

.log
Unable to parse existing variable file
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 31, in loadVariables
    self.allVariables = dict(self.variablefile.items('Variables'))
  File "/usr/lib/python2.7/ConfigParser.py", line 642, in items
    raise NoSectionError(section)
NoSectionError: No section: 'Variables'
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 129, in _read_config
    self.try_load_module(config, section_config.get_name())
  File "/home/pi/klipper/klippy/klippy.py", line 118, in try_load_module
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 64, in load_config
    return SaveVariables(config)
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 20, in __init__
    self.loadVariables()
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 35, in loadVariables
    raise self.gcode.error(msg)
CommandError: 
Unable to parse existing variable file
Internal error during connect: 
Unable to parse existing variable file

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready

dushyantahuja, 2020-05-14:
Try now - I've added some checks to ensure that the file exists, and to create if it doesn't

Goodfeat, 2020-05-14:
Ok

Goodfeat, 2020-05-14:
After the reboot, it worked, but it seems not quite.
I can not find the variable.cfg file now. It seems he is not being created.

How do I know the variables are saved and loaded into a file?
I will try to create a macro with information output

dushyantahuja, 2020-05-14:
Try 
`ECHO VAL={printer.save_variables.<variablename>}`
This should show you if the file has been saved or not.

Goodfeat, 2020-05-14:
[image]

It seems he does not respond to the command

I wrote a macro to display M117 information on the display. it also doesn't display anything

Goodfeat, 2020-05-14:
[image]

I decided to create the file manually and got a similar error

[image]

.log
Unable to parse existing variable file
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 30, in loadVariables
    self.allVariables = dict(self.variablefile.items('Variables'))
  File "/usr/lib/python2.7/ConfigParser.py", line 642, in items
    raise NoSectionError(section)
NoSectionError: No section: 'Variables'
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 129, in _read_config
    self.try_load_module(config, section_config.get_name())
  File "/home/pi/klipper/klippy/klippy.py", line 118, in try_load_module
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 57, in load_config
    return SaveVariables(config)
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 19, in __init__
    self.loadVariables()
  File "/home/pi/klipper/klippy/extras/save_variables.py", line 34, in loadVariables
    raise self.gcode.error(msg)
CommandError: 
Unable to parse existing variable file
Internal error during connect: 
Unable to parse existing variable file

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Goodfeat, 2020-05-14:
I added several variables to this file

Now, in my created macro, the variable is displayed, but for some reason the data on the ECHO command is not displayed VAL = {printer.save_variables.temp}

I think this team does not work. Saving data to a file also does not work. I also found that if the file is empty, a connection error appears. It seems that there is not enough check for the correctness of the file data

dushyantahuja, 2020-05-14:
Let me look at it again. Can you please share your log file, and the variables.cfg files.

Goodfeat, 2020-05-14:
[variable.zip](https://github.com/KevinOConnor/klipper/files/4626924/variable.zip)
I just found this file in the next discussion thread

dushyantahuja, 2020-05-14:
I think I see the issue. You have the following line in your config:

```
save_variable variable = extr VALUE=14
```

This is a gcode that you need execute, not put in the config file. 

Also, the following needs an indent before save_variable:

```
[gcode_macro T0]
gcode = 
save_variable variable = VALUE currenttextruder = extruder #<---- should be indented same as the other gcodes
	SET_GCODE_OFFSET Z=0 MOVE=1
	SET_GCODE_OFFSET X=9.9 MOVE_SPEED=100
	ACTIVATE_EXTRUDER EXTRUDER=extruder
```
Looking at the files - the variables are being saved, and loaded correctly. The file has temp, temp2 and temp1 variables, that are being loaded (from the log file):
```
save_variable state: {'temp2': '23', 'temp1': '4044', 'temp': '123'}
```
Can you try once again - it should work.

Goodfeat, 2020-05-14:
ok i will try this

Goodfeat, 2020-05-14:
I did not update the program code

The printer just freezes when sending a command

```
[save_variables]
filename: ~/variable.cfg

[gcode_macro MES_var]
gcode:
	M301
	SAVE_VARIABLE VARIABLE=temp2 VALUE=14
	M301 
```

[image]

```
save_variable state: {'temp2': '23', 'temp1': '4044', 'temp': '123'}
Declaration of '__main' hides previous menuitem declaration
Declaration of '__filament __load' hides previous menuitem declaration
Declaration of '__filament __unload' hides previous menuitem declaration
Delta max build height 537.37mm (radius tapered above 490.44mm)
Delta max build radius 172.83mm (moves slowed past 156.13mm and 169.29mm)
Extruder max_extrude_ratio=0.266081
Starting serial connect
Loaded MCU 'mcu' 80 commands (v0.8.0-518-g393f37db-20200508_023142-octopi / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1)
MCU 'mcu' config: BUS_PINS_twi1=PB13,PB12 BUS_PINS_twi0=PA18,PA17 STEP_DELAY=2 BUS_PINS_usart0=PA10,PA11,PA17 BUS_PINS_usart2=PB21,PB20,PB24 ADC_MAX=4095 BUS_PINS_usart1=PA12,PA13,PA16 PWM_MAX=255 CLOCK_FREQ=84000000 BUS_PINS_spi0=PA25,PA26,PA27 MCU=sam3x8e STATS_SUMSQ_BASE=256
Sending MCU 'mcu' printer configuration...
Configured MCU 'mcu' (1024 moves)
Starting heater checks for heater_bed
Starting heater checks for extruder
Stats 7196.1: gcodein=5 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=3460 bytes_read=3651 bytes_retransmit=9 bytes_invalid=0 send_seq=170 receive_seq=155 retransmit_seq=2 srtt=0.002 rttvar=0.003 rto=0.025 ready_bytes=481 stalled_bytes=10 freq=83996820 heater_bed: target=0 temp=0.0 pwm=0.000 print_time=4.361 buffer_time=0.229 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 sysload=0.41 cputime=209.921 memavail=654896
Stats 7197.1: gcodein=5 mcu: mcu_awake=0.061 mcu_task_avg=0.000141 mcu_task_stddev=0.000348 bytes_write=4010 bytes_read=3834 bytes_retransmit=9 bytes_invalid=0 send_seq=181 receive_seq=181 retransmit_seq=2 srtt=0.005 rttvar=0.006 rto=0.029 ready_bytes=0 stalled_bytes=0 freq=83999974 heater_bed: target=0 temp=24.7 pwm=0.000 print_time=4.361 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 sysload=0.41 cputime=209.940 memavail=654928
Stats 7198.1: gcodein=5 mcu: mcu_awake=0.061 mcu_task_avg=0.000141 mcu_task_stddev=0.000348 bytes_write=4082 bytes_read=3975 bytes_retransmit=9 bytes_invalid=0 send_seq=186 receive_seq=186 retransmit_seq=2 srtt=0.003 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=83999512 heater_bed: target=0 temp=24.6 pwm=0.000 print_time=4.361 buffer_time=0.000 print_stall=0 extruder: target=0 temp=16.3 pwm=0.000 sysload=0.41 cputime=209.969 memavail=654928
Stats 7199.1: gcodein=5 mcu: mcu_awake=0.061 mcu_task_avg=0.000141 mcu_task_stddev=0.000348 bytes_write=4107 bytes_read=4091 bytes_retransmit=9 bytes_invalid=0 send_seq=189 receive_seq=189 retransmit_seq=2 srtt=0.002 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=83999270 heater_bed: target=0 temp=24.7 pwm=0.000 print_time=4.361 buffer_time=0.000 print_stall=0 extruder: target=0 temp=16.4 pwm=0.000 sysload=0.41 cputime=209.995 memavail=654928
Stats 7200.1: gcodein=5 mcu: mcu_awake=0.061 mcu_task_avg=0.000141 mcu_task_stddev=0.000348 bytes_write=4113 bytes_read=4212 bytes_retransmit=9 bytes_invalid=0 send_seq=190 receive_seq=190 retransmit_seq=2 srtt=0.002 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=83999207 heater_bed: target=0 temp=24.6 pwm=0.000 print_time=4.361 buffer_time=0.000 print_stall=0 extruder: target=0 temp=16.5 pwm=0.000 sysload=0.46 cputime=210.020 memavail=654928
Stats 7201.1: gcodein=10 mcu: mcu_awake=0.061 mcu_task_avg=0.000141 mcu_task_stddev=0.000348 bytes_write=4138 bytes_read=4343 bytes_retransmit=9 bytes_invalid=0 send_seq=193 receive_seq=193 retransmit_seq=2 srtt=0.002 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=83999035 heater_bed: target=0 temp=24.8 pwm=0.000 print_time=4.361 buffer_time=0.000 print_stall=0 extruder: target=0 temp=16.5 pwm=0.000 sysload=0.46 cputime=210.049 memavail=654928
Stats 7202.1: gcodein=10 mcu: mcu_awake=0.002 mcu_task_avg=0.000016 mcu_task_stddev=0.000059 bytes_write=4144 bytes_read=4464 bytes_retransmit=9 bytes_invalid=0 send_seq=194 receive_seq=194 retransmit_seq=2 srtt=0.002 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=83998968 heater_bed: target=0 temp=24.7 pwm=0.000 print_time=4.361 buffer_time=0.000 print_stall=0 extruder: target=0 temp=16.5 pwm=0.000 sysload=0.46 cputime=210.069 memavail=652672

```

dushyantahuja, 2020-05-14:
I'm not sure what the error is - as the log doesn't show anything. Why do you have M301 in your macro - I don't believe that's a recognised gcode.

Goodfeat, 2020-05-14:
code 301 is just a buzzer. it squeaks when the code must be executed. but in this case he does not squeak. the printer just goes wrong when executing code

I tried to insert what you sent me

got another error

[image]

dushyantahuja, 2020-05-14:
try running `save_variable variable=extr value=14` directly in the terminal
on octoprint. I think the additional spaces are not supported by klipper
------------------
Dushyant Ahuja

Goodfeat, 2020-05-14:
[image]

dushyantahuja, 2020-05-14:
looks like you have a space before and after the =

------------------
Dushyant Ahuja

Goodfeat, 2020-05-14:
I just copied your code

dushyantahuja, 2020-05-14:
I'm sorry this is getting so frustrating, and thanks a lot for your help. I
would need the log file to check what's going on.
------------------
Dushyant Ahuja

Goodfeat, 2020-05-14:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4627639/klippy.zip)

dushyantahuja, 2020-05-14:
The log still shows a space between variable and = and extr
Klipper commands follow the format <command name> <parameter>=<value> --
there cannot be a space between <parameter> and the = sign and the <value>

For completeness - can you please get the latest from github, and restart
klipper using `sudo service klipper restart`
------------------
Dushyant Ahuja

Goodfeat, 2020-05-14:
OK

Goodfeat, 2020-05-14:
[klippy2039.zip](https://github.com/KevinOConnor/klipper/files/4627701/klippy2039.zip)

dushyantahuja, 2020-05-14:
Thanks for this - looks like Kevin recently made a lot of changes to the gcode.py file - and somehow, even though I update regularly - these changes are not on my local instance. 

the gcode.py I have - has the function get_str, but the latest on github has a different structure altogether. I'll have to re-write this to make sure it matches the current update that Kevin has made. 

Apologies, it took so long to find this, and thanks for all the help.

Goodfeat, 2020-05-14:
Thanks, I hope that the script will work soon.

P.S There was such a question.
will it be possible to do calculations in this script?

dushyantahuja, 2020-05-14:
> More precisely, not in the script, but in the configuration file variable.cfg?

Not that I know of - all the values are stored as python string literals - so you need to convert them before performing calculations.

Goodfeat, 2020-05-14:
I understood,
to make a calculation, you need to take the current string values, pass them to the script. in a script, convert them to numerical variables and make a calculation. the result is saved again to the configuration file

it seems to sound simple, but in reality I’m not good at python, I am guided only by C and C #. It is sad

dushyantahuja, 2020-05-14:
Same here - this is my first time programming in python.

Goodfeat, 2020-05-14:
For the first time, very good

dushyantahuja, 2020-05-14:
@Goodfeat Do you want to check again - I rebased my local git to 'master' and made changes to the code. It's working on my instance.
Let me know how things go.

Goodfeat, 2020-05-14:
Yes, it seems to work just fine. Both save and load work. Also, the configuration file is created when it is absent.

The only ECHO command VAL = {printer.save_variables.extr} still does not work. But these are trifles. I think that I can somehow register a special macro to output the parameter to the terminal if necessary.

Thank you for your work)

dushyantahuja, 2020-05-15:
Thanks for testing this out and all your patience.
------------------
Dushyant Ahuja

Goodfeat, 2020-05-17:
Hello @dushyantahuja 

There was a question, it seems the script cannot be initialized in the parameter for the screen menu

Could this be due to the fact that the parameters stored in the file are text and not numerical variables?

```
[menu __fast __zoffset]
type: input
name: "Offset Z:{0:05.3f}"
parameter: save_variables.extr|float
input_min: -5
input_max: 5
input_step: 0.005
realtime: true
gcode: 
	M301
#		SET_GCODE_OFFSET Z={0:.3f} MOVE=1
```

```
[menu __fast __zoffset]
type: input
name: "Offset Z:{0:05.3f}"
parameter: save_variables.extr
input_min: -5
input_max: 5
input_step: 0.005
realtime: true
gcode: 
	M301
#		SET_GCODE_OFFSET Z={0:.3f} MOVE=1
```
the variable does not appear on the display. and the format is displayed as a variable name
[image]

dushyantahuja, 2020-05-17:
Possibly - as I had mentioned, all variables are saved as string literals - and you would need to convert them before using them

Goodfeat, 2020-05-18:
I think I did it. I redid your script a bit (it was difficult to do without debug, and I did everything blindly)
I added a new variable to which all converted values are placed
and added some code for the conversion.

Otherwise, it simply does not work to convert the data (the string "parameter:" does not accept any data about the conversion, only pure numbers, even with curly brackets, do not take any value. The conversion does not work with the built-in method |float and |int

Therefore, I supplemented your code with a couple of lines for conversion
`self.list = {}`
```
def loadVariables(self):
        self.allVariables = {}
        try:
            self.variablefile.read(self.filename)
            self.allVariables = dict(self.variablefile.items('Variables'))      
            for var, val in self.allVariables.items():
                try:
                      val = int(val)
                except ValueError:
                      val = float(val)
                self.list.update({var: val})
        except:
            msg = "\nUnable to parse existing variable file"
            logging.exception(msg)
            raise self.gcode.error(msg)

```

```

    def get_status(self, eventtime):
              return self.list

```

maybe there’s an easier option, but I can’t think of anything better

dushyantahuja, 2020-05-18:
Thanks for this - I think you're trying to convert all values to ints or
floats. That won't work for string values, and will probably raise an
error. Will try it out, but I'm keen on keeping this simple, and let people
do the conversion in the gcode macro instead - I'm sure there's a way.

------------------
Dushyant Ahuja

Goodfeat, 2020-05-18:
Maybe you can fix my mistake, I just didn’t see another way how to add verification in the script itself.

I understand that in this way variables with words will not work and will cause an error. But on the other hand, I can’t convert a string variable into a numerical variable by standard means in certain places in the program.

I use it like this for now, because I have only numerical values in the variables.

dushyantahuja, 2020-05-18:
I tried to convert within gcode macros, and used the following macro:

[gcode_macro testmacro]
gcode:
 respond msg="{printer.save_variables.temporary|float + 2.0}"

then ran the following gcodes:

save_variable variable=temporary value=126
testmacro

Received the following response:
Recv: echo: 128.0

Looks like conversion within macros is working. Do you want to try again?
------------------
Dushyant Ahuja

Goodfeat, 2020-05-19:
You understood me wrong. It is not converted specifically for the display menu. Try the same for example for some display menu option
for example for s-offset

try creating a new section for the display menu and place the variable there

```

[menu __main]
type: list
name: Main Menu
items:
    __fast

[menu __fast]
type: list
enable: !toolhead.is_printing
name: Quick setting
items:
	.__zoffset

[menu __fast __zoffset]
type: input
name: "Coating :{0:01.2f}"
parameter: save_variables.zoffsetbed
input_min: -1
input_max: 1
input_step: 0.01
#realtime: true
gcode: 
	SAVE_VARIABLE VARIABLE=zoffsetbed VALUE={0:01.2f}
```

Goodfeat, 2020-05-19:
[image]

the variable from the parameter section is processed by the name section

But the name section is not able to handle python or ninja syntax

dushyantahuja, 2020-05-19:
Aah yes - it won't work like that. Let me see what I can do.
------------------
Dushyant Ahuja

dushyantahuja, 2020-05-24:
@Goodfeat I've made some changes to the module. It now converts values to float, if it can. If not, the values remain as strings. Do you want to test it out?

Goodfeat, 2020-05-24:
Hello, yes I can check it out.
but why only in float type?

is it possible in python to do another check for integers?

```
        for keys in self.allVariables: 
            try:
                self.allVariables[keys] = float(self.allVariables[keys])
            except:
                    try:
                        self.allVariables[keys] = int(self.allVariables[keys])
                    except:
                       self.allVariables[keys] = self.allVariables[keys]
```

dushyantahuja, 2020-05-24:
As far as I could see - conversion to float works for integers as well.
They just come up as 1.0 - is there a use case where that won't work?

------------------
Dushyant Ahuja

Goodfeat, 2020-05-24:
It doesn't seem to work

Goodfeat, 2020-05-24:
Try using what I suggested a few comments earlier, it seems to work fine. you just need to refine it a bit

[15 more replies]