Klipper3d/klipper #2826: Printer screen blank but lit up blue... No info on screen

ZachGreatness, 2020-05-04:
I put the code for the BLTouch used with Teach Tech's tutorial, but now the screen is blank. It was working before, but I unplugged the printer and the Pi then I replugged it and tried to do the BLTouch code and now the screen has nothing on it except the blue light.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4573197/klippy.log)

ZachGreatness, 2020-05-04:
Klipper will not start when trying to restart... It just disconnects. After trying some times, it says "Reported error: File contains parsing errors: /home/pi/printer.cfg"

jakep82, 2020-05-04:
From your log

`ParsingError: File contains parsing errors: /home/pi/printer.cfg
	[line 106]: 'min_point 90, 20\n'`

You missed the colon.  It should be 

`min_point: 90`

closed 2020-05-11 by KevinOConnor