Klipper3d/klipper #2828: TMC configuration example

raugbein, 2020-05-04:
Hi,

I dont added a logfile because at the moment this is a question and not a Issue.

Im using Radds v 1.6 and I want to use the TMC2209 stepper driver for  X, Y, Z. Im searching for a example config for the TMC2209 with stealthChop2.

At the moment I have this on my config file:
```
[stepper_x]
step_pin: ar24
dir_pin: ar23
enable_pin: ar26
step_distance: .0125
endstop_pin: ^ar28
position_endstop: 0
position_max: 330
homing_speed: 50

[stepper_y]
step_pin: ar17
dir_pin: ar16
enable_pin: ar22
step_distance: .0125
endstop_pin: ^ar30
position_endstop: 0
position_max: 310
homing_speed: 50

[stepper_z]
step_pin: ar2
dir_pin: ar3
enable_pin: ar15
step_distance: .0125
endstop_pin: ^ar32
position_endstop: 0.5
position_max: 400
```

I searched a lot in the Internet, but is hard to find klipper config for the TMC2209. I also searched in the config folder for examples, but also nothing found.

Can you give me a example configuration?
Is soldered like this picture 
[image]

Hywelmartin, 2020-05-04:
https://github.com/KevinOConnor/klipper/blob/9a92b346fd81386c54464b45d8310dab972b6bbd/config/example-extras.cfg#L1441

raugbein, 2020-05-04:
Excuse me for asking such stupid questions. I am new to Klipper and stepper drivers and don't want to see everything in flames.

First I add this additional after [stepper_x] block
```
[tmc2209 stepper_x]
uart_pin: ar27   //Radds AUX1 27
tx_pin: ar25   //Radds AUX1 25
microsteps: 16
interpolate: True
run_current: 1.5  // how can I calculate this value? Can I add 1.5A from Nema17 description?
hold_current: 0.5 //  What is the minimum/hold best practice?
sense_resistor: 0.110
stealthchop_threshold: 60  //60mm/s, I dont know which value is a good practice
```

But I have only 7 PINS on AUX1, I need for 5 steppers 10.  I dont use a display on the Radds, I have my Display on the Raspberry. Can I use some PINS from the 'Display and Encoder' connector?

[image]

[image]

Is this configuration and PINs are correct?

Hywelmartin, 2020-05-05:
Normally you only need UART-pin, so 5 pins as max.. The 2209 has addressable controllers so for 4 stepperdrivers it is possible to only use one pin on the MCU.. (2 for 5-8 steppers)

raugbein, 2020-05-05:
> Normally you only need UART-pin, so 5 pins as max.. The 2209 has addressable controllers so for 4 stepperdrivers it is possible to only use one pin on the MCU.. (2 for 5-8 steppers)

Ok, does this mean the example from above is correct?

Hywelmartin, 2020-05-05:
[tmc2209 stepper_x]
uart_pin: ar27
microsteps: 16
interpolate: True
run_current: 0.8 
hold_current: 0.5
stealthchop_threshold: 60 # 0 or 200

raugbein, 2020-05-06:
Hi,
I added only one Stepper Driver at this moment.

in printer.cfg, I have
```
[stepper_x]
step_pin: ar24
dir_pin: ar23
enable_pin: ar26
step_distance: .0125
endstop_pin: ^!ar28
#endstop_pin: ^ar34
position_endstop: 0
position_max: 330
homing_speed: 50
#homing_retract_dist: 0

[tmc2209 stepper_x]
uart_pin: ar27
microsteps: 16
interpolate: True
run_current: 0.8
hold_current: 0.5
stealthchop_threshold: 60 # 0 or 200
```

Wee I click to X Home, I get this error:
`!! Failed to home x: Timeout during homing`

[klippy.log](https://github.com/KevinOConnor/klipper/files/4588393/klippy.log)

KevinOConnor, 2020-05-11:
The enable pin should be inverted (eg, "!ar26") on nearly all drivers (including the Trinamic).

-Kevin

closed 2020-05-24 by KevinOConnor