Klipper3d/klipper #2831: [FEATURE] Analyse whole Gcode fro errors before print to avoid killing print

Benraay, 2020-05-05:
Hi I am using Klipper for a few days now I was very impressed with this firmware a lot of interesting things But !

Could you please analyse Gcode for impossible out of range moves or max volumetric extrusion errors I failed 3 prints because of an error I did'nt manage, max volumetric speed seems to be an S3D wrong gcode, OK but why not triggering an error before the print happens it will avoid killing a long time print !

I suppose this is possible, like you do it live, it will be OK for most people waiting a bit before print giving the time to analyse gcode !

KevinOConnor, 2020-05-08:
Such a feature would certainly be useful.  An interested developer would need to implement it and submit it.

-Kevin

Benraay, 2020-05-08:
on marlin there are some things tested at compile time like out of bed moves

DanielJoyce, 2020-05-19:
Yeah, it would be nice to scan to check for the over-extrusion warning that seems to kill prints sometimes.

KevinOConnor, 2020-06-14:
I'm going to mark this as inactive until it is actively under development.

-Kevin

closed 2020-06-14 by KevinOConnor