Klipper3d/klipper #2832: Fysetc Mini12864 Panel on ramps 1.4 with Klipper

parapake, 2020-05-05:
I have a ramps 1.4 with klipper, octoprint,  with RepRapDiscount 128x64 Full Graphic is all ok
I have seen that klipper now supports the Mini12864 Panel and I change it, I have configured it as klipper says but only the button light works  and the screen remains dark, I rotate 180 the lcd connectors on my ramps, Is it a contrast problem? can you help me?
Thanks
[printer.txt](https://github.com/KevinOConnor/klipper/files/4582020/printer.txt)

parapake, 2020-05-05:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4582125/klippy.log)

Hywelmartin, 2020-05-06:
have you looked here??

https://wiki.fysetc.com/Mini12864_Panel/

jalanjarosz, 2020-05-06:
I have had this display working on my Re-ARM board on my  bench.

Here is the config I'm using:

```
######################################################################
# MKS Mini 12864 LCD v2.1 with Neopixel
######################################################################

# Make sure that the EXP1 and EXP2 are rotated correctly on the
# display board. The cutouts on the connectors should be towards the
# center of the PCB. See:
#   https://reprap.org/wiki/MKS_MINI_12864#Physical_Interface
# If they are wrong, the connector housing can be pried off carefully
# with a small screwdriver and relocated the correct way.

# Re-ARM board does not have the option to designate Hardware SPI.
# SPI pins were wired from Re-ARM headers to display as follows:
# Fysetc MKS Mini v2.1 Pin ~ Re-ARM MCU Pin ~ Re-ARM Header ~ RAMPS v1.4.4 pin
# EXP2-1 ~ P0.17 ~ J3-6 / MISO ~ D50
# EXP2-2 ~ P0.15 ~ J3-9 / SCK ~ D52
# EXP2-6 ~ P0.18 ~ J3-10 / MOSI ~ D51
# Fysetc MKS Mini Display v2.1 works on Re-ARM when wired with the following

# Support for a display attached to the micro-controller.
#[display]
#lcd_type: uc1701
#cs_pin: EXP1_3
#a0_pin: EXP1_4
#rst_pin: EXP1_5
#contrast: 63
#encoder_pins: ^EXP2_3, ^EXP2_5
#click_pin: ^!EXP1_2
#kill_pin: ^!EXP2_8

#[neopixel mks_display]
#pin: EXP1_6
#chain_count: 3
#color_order_GRB: False
#initial_RED: 0.4
#initial_GREEN: 0.0
#initial_BLUE: 1.0

######################################################################
# Plug pin locations
######################################################################

# Some micro-controller boards and displays use inconsistent labeling
# for the EXP1 and EXP2 headers. The following diagram shows the
# correct location of pin 1 along with ground and power pins on the
# EXP1 and EXP2 plugs:
#
#          EXP1:                        EXP2:
#   +-----------------+          +-----------------+
#   |  o  o  o  o  5V |          |  o  o  o  o  o  |
#   |  1  o  o  o GND |          |  1  o  o  o GND |
#   +------     ------+          +------     ------+
#
# Some boards may have the cutout in the wrong location. If so, it may
# be possible to carefully pry the plastic header off of the pins with
# a small screwdriver, then correct the orientation and reapply the
# plastic header.
```

closed 2020-05-08 by parapake