Klipper3d/klipper #2834: probe offsets correct but when BED_MESH_CALIBRATE is ran the "probe xxx.xx" is incorrect

iH8teF1ames, 2020-05-05:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4583563/klippy.log)
[image]

iH8teF1ames, 2020-05-05:
when looking at the terminal output it says the probe should be at  48.x but its pretty far off. The X and Y offset are from the BLTouch mounts page so they should be correct.

Arksine, 2020-05-05:
The terminal doesn't tell you where the probe is, it tells you where the nozzle is.    It appears that you recently updated Klipper, so you may not be aware that the behavior of bed_mesh has changed (as outlined in Config_Changes.md).   With the current version of bed_mesh, it will generate points that place probe the actual designated area,  I see a `mesh_min: 10, 25` in your config, and it appears that your probe may be around X=10 in that picture.

iH8teF1ames, 2020-05-05:
Interesting, ill look into that. So even though the terminal output says "probe at 48.x" it doesn't actually mean the probe and the position that its outputting is where the nozzle is?

Arksine, 2020-05-05:
Correct.  When it says "probe at 48.x" the coordinate is the current tool position.  Perhaps that is something we could remedy by showing both the probe and tool locations.

iH8teF1ames, 2020-05-05:
I think that could save some confusion for the future. Thank you for the insight! I will close the issue.

closed 2020-05-05 by iH8teF1ames