Klipper3d/klipper #2835: Home location changes after initial homing operation

w1ebr, 2020-05-06:
Recv: // Klipper state: Disconnect
[...]
Recv: // Klipper state: Ready
Send: Status
Recv: // Klipper state: Ready
Recv: ok
[...]
Send: G91
Recv: ok
Send: G28 X0 Y0
Recv: ok
Send: G90
Recv: ok
[...]
Send: get_position
Recv: // mcu: stepper_x:12192 dual_carriage:27544 stepper_y:-40815 stepper_z:658
Recv: // stepper: stepper_x:111.500000 dual_carriage:311.350000 stepper_y:100.000000 stepper_z:1.480000
Recv: // kinematic: X:111.500000 Y:100.000000 Z:1.480000
Recv: // toolhead: X:111.500000 Y:100.000000 Z:1.480000 E:0.000000
Recv: // gcode: X:111.500000 Y:100.000000 Z:1.480000 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
[...]
Send: G28 X0 Y0
Recv: ok
[...]
Send: get_position
Recv: // mcu: stepper_x:16042 dual_carriage:27541 stepper_y:-40808 stepper_z:688
Recv: // stepper: stepper_x:150.000000 dual_carriage:311.350000 stepper_y:100.000000 stepper_z:1.480000
Recv: // kinematic: X:150.000000 Y:100.000000 Z:1.480000
Recv: // toolhead: X:150.000000 Y:100.000000 Z:1.480000 E:0.000000
Recv: // gcode: X:150.000000 Y:100.000000 Z:1.480000 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok

[klippy.log](https://github.com/KevinOConnor/klipper/files/4584465/klippy.log)

theophile, 2020-05-08:
Following, as I've had this issue for a long time. Sometimes if I home Z again immediately after homing it, it will do the homing moves without actually engaging the endstop the second time. It'll come right up to it and stop, but the switch never clicks. There doesn't seem to be a rhyme or reason to it.

KevinOConnor, 2020-05-08:
It's not clear to me what is being reported here.  I recommend explicitly describing "what you saw" and "what you expected to see".  Also, please attach run M112 immediately after the unexpected event and attach that log file here.

-Kevin

w1ebr, 2020-05-08:
Hello Kevin,
I expected to see a get_position issued immediately  after  a g28 to always report the same location. You can see that it did not report the same position. This makes me wonder if the g91 command preceding the first g28 affects its behavior.

KevinOConnor, 2020-05-11:
Unfortunately, without a log from the event there is little that can be done.

-Kevin

w1ebr, 2020-05-11:
Hi Kevin, the log is attached to the OP.

KevinOConnor, 2020-05-21:
We'll need a log from the event - see https://github.com/KevinOConnor/klipper/issues/2835#issuecomment-625864930

-Kevin

closed 2020-06-14 by KevinOConnor