Klipper3d/klipper #2837: Using two endstops (probe & physical switch)

sedlacekdavid, 2020-05-06:
Hi Guys, is it possible to use both sensors at the same time for a single endstop? Like keep bltouch as the main Z probe, but have a Z axis endstop switch as a sort of fail safe mechanism, if bltouch fails and crashes to the bad (happened to me several times)

Thank you, David

theopensourcerer, 2020-05-06:
I have a Piezo probe for levelling _and_ a Z endstop on minimum so I should think so, yes.

sedlacekdavid, 2020-05-06:
@theopensourcerer OK and does your probe also acts as a Z endstop? That's what I've forgot to mention. I'm using the probe also as and end stop. But I'd like the switch to kick in if needed.

theopensourcerer, 2020-05-06:
@sedlacekdavid No - it's a probe. I have an endstop and a probe. They are different. 

I'm not sure if you can use a probe like that; for both probing & endstop _and_ have an endstop as well. Sorry.

KevinOConnor, 2020-05-11:
It's possible to define a switch for homing and a probe for probing.  There is currently no way to home or probe using both tools.

-Kevin

KevinOConnor, 2020-05-24:
I'm going to close this as it looks like the original question was answered.

-Kevin

closed 2020-05-24 by KevinOConnor

Averjek, 2020-09-27:
> It's possible to define a switch for homing and a probe for probing. There is currently no way to home or probe using both tools.
> 
> -Kevin

Hi,it also possible for two endstops on Z axis?
I tried to configure bltouch(3dtouch) with 2 motors and 2 endstops working separately ,with probe only for bed mesh,but it failed