Klipper3d/klipper #2838: Printer X axis goes out of bounds, ignores limit switch.

eringuet, 2020-05-06:
Hi, I've printed this model without issue. When I tried to print it again, the X axis went out of bounds both on the left and right side, ignored the limit switches and crashed on the edges.

I've attempted to upgrade to the latest build of Klipper, updating the firmware.bin file on my skr mini e3 as well. I've updated Cura to 4.6.1. Octoprint is also running the latest version. I've tightened my z axis belt. Printer is an Ender 3 pro. 

I tried turning the model around, re-slicing, changing the bed size, set the stealthchop to 9999. 

[klippy-issue.log](https://github.com/KevinOConnor/klipper/files/4588857/klippy-issue.log)

eringuet, 2020-05-06:
[image]

It did it again with a completely different model. I also tried with Prusaslicer and had the same issue. 
[klippy-issue2.log](https://github.com/KevinOConnor/klipper/files/4589878/klippy-issue2.log)

BlackStump, 2020-05-07:
try changing your stealthchop to 0
9999 is not a sane figure
it seems you a skipping steps

eringuet, 2020-05-07:
Someone on discord asked me to try to troubleshoot. I had X and Y at 250 and Z at 5 prior to that. I'll see if 0 helps. Thanks.

eringuet, 2020-05-07:
I've reinstalled everything from scratch and picked tag v0.8.0 to see if the problem was introduced after my last upgrade. Now I see this in my terminal: "Unable to read tmc uart 'stepper_x' register IFCNT". Does that mean the stepper driver is defective?

eringuet, 2020-05-08:
I've switched the X and Y and the problem went to the Y axis. I think my SKR mini E3 might be defective. Probably safe to close the issue. Thanks.

closed 2020-05-24 by KevinOConnor