Klipper3d/klipper #2842: Bed Visualizer plugin

Tomkee, 2020-05-07:
After updating Klipper I have problem with Bed Visualizer for plugin for OctoPrint. It used to run just fine before update with this GCODE commands

G29
@BEDLEVELVISUALIZER
BED_MESH_OUTPUT

Now after whole proces plugin don't update Bed visualisation its just run out of time (no mather how log I will let it run). Do I miss something? Do I need to add something?

This is my Klipper Confuguration (after changes I made to work with newest update)

[stepper_x]
step_pin: PD7
dir_pin: !PC5
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC2
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PC6
dir_pin: !PC7
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC3
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PB3
dir_pin: PB2
enable_pin: !PA5
step_distance: .0025
endstop_pin:probe:z_virtual_endstop
position_max: 250
position_min:-9

[extruder]
max_extrude_only_distance: 800.0
max_extrude_cross_section: 60.0
step_pin: PB1
dir_pin: !PB0
enable_pin: !PD6
step_distance: 0.010315
nozzle_diameter: 0.400
filament_diameter: 1.750
pressure_advance: 0.6
pressure_advance_lookahead_time: 0.010
heater_pin: PD5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA7
#control: pid
# tuned for stock hardware with 200 degree Celsius target
#pid_Kp: 21.527
#pid_Ki: 1.063
#pid_Kd: 108.982
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PD4
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA6
control: pid
# tuned for stock hardware with 50 degree Celsius target
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PB4

[mcu]
serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AC00FRRN-if00-port0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[display]
lcd_type: st7920
cs_pin: PA3
sclk_pin: PA1
sid_pin: PC1
encoder_pins: ^PD2, ^PD3
click_pin: ^!PC0

[bltouch]
sensor_pin: ^PC4
control_pin: PA4
x_offset: -40
y_offset: -15
#z_offset: -3.710
pin_move_time: 0.4
pin_up_touch_mode_reports_triggered: False

[bed_mesh]
speed: 80
horizontal_move_z: 5
mesh_min: 20, 30
mesh_max: 195, 210
probe_count: 5,5
fade_start: 1.0
mesh_pps: 2,2

[gcode_macro G29]
gcode:
    BED_MESH_CALIBRATE

[safe_z_home]
home_xy_position: 155,120
speed: 80.0
z_hop: 10.0
z_hop_speed: 10.0

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bltouch]
#*# z_offset = 1.400
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 27.978
#*# pid_ki = 1.651
#*# pid_kd = 118.556
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	  -0.080000, -0.075000, 0.002500, 0.030000, 0.052500
#*# 	  0.020000, 0.027500, 0.042500, 0.055000, 0.030000
#*# 	  0.015000, 0.012500, 0.010000, 0.062500, 0.047500
#*# 	  0.042500, 0.067500, 0.057500, 0.087500, 0.067500
#*# 	  0.070000, -0.007500, 0.035000, 0.022500, 0.000000
#*# x_count = 5
#*# y_count = 5
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
#*# min_x = 20.0000000001
#*# max_x = 195.0
#*# min_y = 30.0
#*# max_y = 210.0

Tomkee, 2020-05-08:
[Log.zip](https://github.com/KevinOConnor/klipper/files/4598125/Log.zip)

Arksine, 2020-05-08:
Unfortunately none of the attached logs show your attempt to run BED_MESH_CALIBRATE.  As far as I know nothing has changed with BED_MESH_OUTPUT.  I can't speak as to why the bed visualizer plugin isn't responding correctly.

Lefuneste83, 2020-05-13:
I've had the same issue yesterday, the BED_MESH_CALIBRATE command would run fine, BED_MESH_OUTPUT also. No errors at all in the terminal, but the plugin would just stay waiting to display data until timeout occurs. I have unlocked this by entering manually the commands @BEDLEVELVISUALIZER
BED_MESH_OUTPUT
manually in the terminal once the probing is finished and before the timeout occurs. Strangely enough, this not only unlocks the display of the concurrent plugin task, but also seems to make it work even afterwards. All is not perfect and I had this issue once more after. I have no idea why the plugin sometimes seems to be unable to get the data from the probing, and why sending the exact same command manually unlocks the situation.

KevinOConnor, 2020-06-01:
Closing pending logs from the event.

-Kevin

closed 2020-06-01 by KevinOConnor