Klipper3d/klipper #2843: BLTouch (w/ SKR v1.4 turbo + TMC2209

bgbnbigben, 2020-05-08:


fxbisto, 2020-05-10:
You can make sure the board and configuration side of things is working by connecting the Z endstop to a mechanical switch instead of the BLTouch. Send QUERY_ENDSTOPS in the terminal to see if Z reports triggered or open, then manually trigger the switch and issue the command again.

Just to clarify, the expected responses are:
z:open - without connecting a switch or switch not depressed
z:TRIGGERERED - when switch is depressed

If yours doesn't behave like this it's likely there's an error in pin configuration for the Z endstop or hardware issue with the board itself. If it _does_ work fine then the issue likely lies with the BLTouch itself.

Let us know how you get on :)

KevinOConnor, 2020-05-11:
I recommend you follow the directions at https://www.klipper3d.org/BLTouch.html .  If you can not get the "sensor pin test" to pass, then it is an indication of an incorrectly configured sensor_pin or a wiring/hardware failure.

-Kevin

bgbnbigben, 2020-05-12:
Does it count as a hardware flaw if it's by design? ^_^

Notably with this board and *specifically* the 2209 drivers, there's no jumper for the 2209 DIAG pin; that means to test a physical endstop/mechanical switch, I need to cut the pin (or desolder it, or bend it or otherwise mutilate the pin). To use sensorless homing, OTOH, the DIAG pin needs to not be connected to the <axis>-min pins.

Of course, in theory the probe pin of the board has the DIAG pin wired correctly but it's BTT who knows.

In this particular case I was able to get by with forcing the hardware pullup of the sensor pin, i.e. `sensor_pin: ^P0.10`. With that the probe/board seems to report the correct values.

closed 2020-05-12 by bgbnbigben