Klipper3d/klipper #2844: Can I add support for common 12864 displays directly?

Goodfeat, 2020-05-08:
Hello.

I have a 12864 CD display for Ramps 1.4. but as the source found, support is not directly implemented.

I tried to add my configuration for the clipper.
she looks like this

```
[display]
lcd_type: st7920
rs_pin: EXP1_4
e_pin: EXP1_3
d4_pin: EXP1_5
d5_pin: EXP1_6
d6_pin: EXP1_7
d7_pin: EXP1_8
encoder_pins: ^ EXP2_3, ^ EXP2_5
click_pin: ^! EXP1_2
#kill_pin: ^! EXP2_8

[output_pin beeper]
pin: EXP1_1

[board_pins]
aliases:
     # Common EXP1 header found on many "all-in-one" ramps clones
EXP1_2 = ar40, EXP1_4 = ar63, EXP1_6 = ar50, EXP1_8 = ar53, EXP1_10 = <5V>,
EXP1_1 = ar62, EXP1_3 = ar64, EXP1_5 = ar48, EXP1_7 = ar52, EXP1_9 = <GND>,
 
     # EXP2 header
EXP2_3 = ar44, EXP2_5 = ar42
```
Is it possible to make the display work directly? Can I add support for this display?

Goodfeat, 2020-05-08:
attached log file
[klippy.log](https://github.com/KevinOConnor/klipper/files/4597762/klippy.log)

Hywelmartin, 2020-05-08:
With the right pins it would work I think.. 

[image]

Goodfeat, 2020-05-09:
@Hywelmartin Unfortunately, this will not even theoretically work. Of course I will try, but something tells me that this pinout does not match the pinout of my display reprap.

This is the standard pinout for the 12864 display REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

[image]

even after looking at the source code, it’s already clear that connecting this display is not possible, because it is not provided for in software
[image]

However, 2004 text displays can be connected in this way. This can be seen from the examples, as well as if you look at the source code for HD44780

[image]

The display is common, and I was hoping there would be no problems connecting it. However, in practice, nothing is displayed.

I beg the developer to help with the implementation of the connection of this display.

Please add display support through pins 'rs', 'e', ​​'d4', 'd5', 'd6', 'd7' for REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Goodfeat, 2020-05-09:
Well, I still figured out how the display works. It turns out he has two interfaces, and you can connect it by cs_pin signals sclk_pin sid_pin
I tried connecting as follows

[display]
lcd_type: st7920
cs_pin: ar63
sclk_pin: ar48
sid_pin: ar64
encoder_pins: ^ ar44, ^ ar42
click_pin: ^! ar40

[output_pin beeper]
pin: ar62

and it worked))

closed 2020-05-09 by Goodfeat