Klipper3d/klipper #2845: Manual probe failed! Use TESTZ commands to position the nozzle prior to running ACCEPT.

Hairyloon, 2020-05-08:
I suspect this may relate to my other issue: https://github.com/KevinOConnor/klipper/issues/282
But in short, I try to run the Z_ENDSTOP_CALIBRATE, move it up and down a bit with testz, but when I accept, it says:
"Manual probe failed! Use TESTZ commands to position the nozzle prior to running ACCEPT."

[klippy.log](https://github.com/KevinOConnor/klipper/files/4598077/klippy.log)

fxbisto, 2020-05-10:
I'm not certain that this relates to your other issue as I have a brand new setup using a fairly stock config/generic_bigtree_skr_v1.4.cfg (I've added a bed_mesh section to printer.cfg) which appears to behave the same:
 
```
Send: testz z=-.1 
Recv: // Z position: ?????? --> -0.650 <-- -0.550 
Recv: ok
 
Send: testz z=+.1 
Recv: // Z position: -0.650 --> -0.550 <-- -0.450 
Recv: ok
 
Send: testz z=- 
Recv: // Z position: -0.650 --> -0.600 <-- -0.550 
Recv: ok
 
Send: testz z=+ 
Recv: // Z position: -0.600 --> -0.575 <-- -0.550 
Recv: ok
 
Send: testz z=+.1 
Recv: // Z position: -0.550 --> -0.475 <-- -0.450 
Recv: ok
 
Send: testz z=- 
Recv: // Z position: -0.550 --> -0.512 <-- -0.475 
Recv: ok
 
Send: testz z=- 
Recv: // Z position: -0.550 --> -0.531 <-- -0.512 
Recv: ok
 
Send: accept 
Recv: // Manual probe failed! Use TESTZ commands to position the 
Recv: // nozzle prior to running ACCEPT. 
Recv: ok
```

Not certain which git hash the Pi was using previously but I'm fairly sure I'd used this command numerous time on previous builds.

Current build:
Author: Kevin O'Connor <kevin@koconnor.net>
Date:   Tue May 5 13:57:07 2020 -0400

    klippy: Minor code movement
    
    Move set_rollover_info() - no code changes.
    
    Signed-off-by: Kevin O'Connor <kevin@koconnor.net>

commit 393f37dbfcae858e38aa119fb32a61ae3239c57d

KevinOConnor, 2020-05-11:
During a manual probe, or any tool that uses it, you must always _descend_ towards the bed.  So, don't start the tool when the endstop is already triggered, and make sure the "xy move height" for the tool (if applicable) is configured to be well above the bed.

-Kevin

Hairyloon, 2020-05-14:
> During a manual probe, or any tool that uses it, you must always _descend_ towards the bed.

That much is obvious, but during the process in order to find the limit, we will often go past it and back off.
Are you saying that the setting can only be saved after a downwards move?
If so, then this ticket can be closed, but could the error be changed to reflect that information? 
It might make things easier.

KevinOConnor, 2020-05-24:
It is not necessary to end on a move that descends.  The final Z position must only be lower than the start Z position.

-Kevin

Hairyloon, 2020-05-24:
> It is not necessary to end on a move that descends. The final Z position must only be lower than the start Z position.
> 
> -Kevin

OK, that seems pretty clear. Thank you.

closed 2020-05-24 by Hairyloon

jamon, 2021-07-10:
Sorry to resurrect this dead thread, but I'm struggling with this.  If the final Z position must be lower than the start Z position, how are you supposed to ever move it back up if you've accidentally set it too far?  Is there a command to manually set it, or to reset it?

Hairyloon, 2021-07-10:
Abort the calibration, move the head up, then start the calibration afresh.